alaudet/raspi-sump #6: Gmail now require oauth2 to send email notifications

alaudet, 2016-02-23:
If you are using gmail smtplib throws up an exception because google considers username/password less secure.

The following traceback is provided when running rsump.py directly  (this example used the latest beta version of python3 for raspisump which has not yet been released but the same issue will a happen with v0.8.0

```
pi@raspidev:~/raspi-sump $ python3 /usr/local/bin/rsump.py
Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 23, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.4/dist-packages/raspisump/reading.py", line 67, in water_depth
    alerts.determine_if_alert(water_depth)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/alerts.py", line 142, in determine_if_alert
    smtp_alerts(water_depth)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/alerts.py", line 108, in smtp_alerts
    server.login(username, password)
  File "/usr/lib/python3.4/smtplib.py", line 652, in login
    raise SMTPAuthenticationError(code, resp)
smtplib.SMTPAuthenticationError: (534, b'5.7.14 <https://accounts.google.com/ContinueSignIn?sarp=1&scc=1&plt=AKgnsbuhF\n5.7.14 Ek2...3EA> Please log in via your web browser and\n5.7.14 then try again.\n5.7.14  Learn more at\n5.7.14  https://support.google.com/mail/answer/78754 xh7sm11463113igb.1 - gsmtp')
```

I will look to add oauth2 to raspi-sump but in the meantime you can relax the setting in gmail at the following address after logging into your gmail account via the web interface.

Go to:  
https://www.google.com/settings/security/lesssecureapps

and select Turn On  (to allow less secure apps to access your gmail account)

I recommend creating a throw away account in gmail for these notifications so not to enable this on your main account.  In any event a good secure password should be fine and the world will not end.

Oauth2 is now on my radar for raspi-sump

spankurmonkey, 2016-04-27:
I am not able to get it installed.  I followed the steps but when I try the sudo rsump.py nothing happens.  if i check the bin folder the rsump.py file is not there.  However, the raspi-sump directory is found at /home/pi ....

alaudet, 2016-04-28:
Hi please open a new issue.
If rsump.py is not in your /usr/local/bin folder but you have a /home/pi/raspi-sump folder there then it was uninstalled for whatever reason.

Rerun the setup by using pip.

If you originally install for Python3  run   sudo pip3 install raspisump
If you originally installed with Python2 run sudo pip install raspisump

If it complains that it already exists you can try sudo pip install -U raspisump

alaudet, 2016-05-03:
I am going to close this issue.  I have no plans of adding Oauth2 at the moment just for gmail.  Gmail offers an alternative already to relax the Oauth2 requirement.

I recommend using a throw away GMail account if lack of Oauth2 concerns you.  Or use another email provider.  That's not to say it will not be added in the future at some point, but I will need a better reason than Google's decision to use Oauth2 as the default for non webmail clients.

alaudet, 2023-06-17:
GMAIL now uses App passwords.  Office365 also implements this feature.

closed 2023-06-17 by alaudet