alaudet/raspi-sump #8: error

morejava, 2016-04-27:
When I run rsump.py I receive this error.  Suggestions?

:~ $ sudo rsump.py
Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 26, in <module>
    reading.water_depth()
  File "/usr/local/lib/python2.7/dist-packages/raspisump/reading.py", line 66, in water_depth
    alerts.determine_if_alert(water_depth)
  File "/usr/local/lib/python2.7/dist-packages/raspisump/alerts.py", line 125, in determine_if_alert
    smtp_alerts(water_depth)
  File "/usr/local/lib/python2.7/dist-packages/raspisump/alerts.py", line 96, in smtp_alerts
    server = smtplib.SMTP(configs['smtp_server'])
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 316, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
    return socket.create_connection((host, port), timeout)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused

alaudet, 2016-04-28:
Hi this looks like an SMTP (email) error.  What values did you give in raspisump.conf for for the email server?  It looks like you are trying to send from localhost but you do not have a local smtp server on your pi.

morejava, 2016-04-28:
I'm using gmail. tls=1 STMP=1 port 587

-- 
​​
Andy Stumpf
Director of High School Ministries

Christ Lutheran Church
4325 Sumner
Lincoln, NE  68506

Office  402-483-7774 ext. 109
Mobile
​/Text 402-840-1708

www.christlutheranchurch.org

alaudet, 2016-04-28:
Please email me the output of

`find /usr | grep "raspisump"`

Also email me the entire [email] section of your raspisump.conf file.  Just remember to remove the password if you are using authentication.

Did you relax the restrictions on your gmail account as well.  Gmail requires oauth2 unless you tell it not to use it.  

See the issue tracker https://github.com/alaudet/raspi-sump/issues/6 for details.  It doesn't look like that error but its still something to consider.

morejava, 2016-04-29:
*pi@Home*:*~ $* find /usr | grep "raspisump"

/usr/local/lib/python3.4/dist-packages/*raspisump*

/usr/local/lib/python3.4/dist-packages/*raspisump*/webchart.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/alerts.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/reading.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/__init__.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/checkpid.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/log.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/todaychart.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/ttt.py

/usr/local/lib/python3.4/dist-packages/*raspisump*/__pycache__

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/checkpid.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/webchart.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/todaychart.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/reading.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/alerts.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/ttt.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/__init__.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*
/__pycache__/log.cpython-34.pyc

/usr/local/lib/python3.4/dist-packages/*raspisump*-1.0.1.egg-info

/usr/local/lib/python3.4/dist-packages/*raspisump*
-1.0.1.egg-info/installed-files.txt

/usr/local/lib/python3.4/dist-packages/*raspisump*-1.0.1.egg-info/PKG-INFO

/usr/local/lib/python3.4/dist-packages/*raspisump*
-1.0.1.egg-info/SOURCES.txt

/usr/local/lib/python3.4/dist-packages/*raspisump*
-1.0.1.egg-info/requires.txt

/usr/local/lib/python3.4/dist-packages/*raspisump*
-1.0.1.egg-info/dependency_links.txt

/usr/local/lib/python3.4/dist-packages/*raspisump*
-1.0.1.egg-info/top_level.txt

*p*

morejava, 2016-04-29:
# Configuration file for Raspi-Sump

# All configurations should be done in this file.

# This file should be located in the /home/pi/raspi-sump directory

[gpio_pins]

# Raspi-Sump uses the actual GPIO numbers and not the Pi numerical pin

# numbers.  You can use any available GPIO's for Trig and Echo.

# Trig pin sends the signal.  Value is the number of the GPIO

# on the RaspberryPi

trig_pin = 17

# Echo pin receives the signal.  Value is the number of the GPIO

# on the RaspberryPi

echo_pin = 27

[pit]

# Unit of measure.  You may choose to use metric or imperial measurements.

# The unit variable must be set as metric or imperial and is case sensitive.

# Any other setting will cause an error.

unit = imperial

# critical_water_level is the water level in the sump pit at which point an

# email alert should be triggered.

# If unit = metric then critical_water_level = centimeters

# If unit = imperial then critical_water_level = inches

critical_water_level = 4

# The pit_depth is the distance between the sensor and the bottom of the

# sump pit.

# Like critical_water_level, enter centimeters for metric and inches for

# imperial

pit_depth = 27

# rsump.py should be run with a cron job.  However cron is limited to

# 1 minute as its minimum interval. If you need to monitor the sump level

# more frequently you can set reading_interval to the desired time in
seconds.

# Default is 0 meaning only one reading will occur and the program will
exit.

# If set to 0 call the script with a cron job and select your interval
there.

# If setting to a value other than 0 you should run rsumpmonitor.py from a
cron

[... 287 lines cut ...]
# example -- email_to = cellnumber@wireless_carrier (for sms email alerts)

# You can also use a regular email address

# To add multiple recipients seperate email addresses with a comma.

# e.g. Adding a single recipient

#   email_to = single_email@somewhere.com

#

# e.g. Adding multiple recipients

#   email_to = recipient1@somewhere.com, recipient2@somewhere.com

email_to = sump.pump@gmail.com

# Notification will be coming from the following address.

# You can enter your email address here as the sender.

# example  email_from = Raspi-Sump <youremail@yourprovider.com>

email_from = andy.stumpf77@gmail.com

alaudet, 2016-04-29:
The format is all screwed up on the file.  Any way to attach it?

A couple of issues

smtp_server = 587     is wrong.  This line needs to be   smtp_server =
smtp.gmail.com:587

Also reading interval is set to take a reading every 1 second, I don't
think you want that.  This needs to be 0 and run through a cron job
every minute if that is what you are trying to do.

Set this to 0 and reboot to make sure all rsump.py processes are dead.

Setting it to 1 will cause issues with the process trying to start
before the last one stopped.

Start there and then reassess.

morejava, 2016-04-29:
NICE!!!! great suggestions and I can't believe i missed the "localhost"
thing *sigh*

Almost all is running... just need to figure out the chart now...

alaudet, 2016-04-29:
Since  this has solved the connection problem I will close this issue.

closed 2016-04-29 by alaudet