alaudet/raspi-sump #9: Missing Web Page Functionality.

spankurmonkey, 2016-05-03:
I have the raspi-sump up and running...

The issue I have is seeing historical days - like what is shown here: https://raspisump.linuxnorth.org/todaychart/

I only have a link that says "past water readings" but when clicked just goes to a directory .

Seems if there is missing html pages/coding.. when comparing to : https://raspisump.linuxnorth.org/todaychart/

alaudet, 2016-05-03:
The web functionality with raspisump is just a bare bones view of past water readings.  The graphing you see on my web page is not part of raspisump. Those graphs are uploaded to an offsite webserver and have some programming behind them to display monthly graphs.

On the raspberry pi itself I just wanted something very bare bones that someone could see what's up with their graphs and also have access to older ones.  

Making the web server display something nicer is something I would like to do, or hopefully one day a good web designer would use raspisump and see fit to contribute that part.  :-)

closed 2016-05-03 by alaudet

budpi, 2017-02-18:
What external/offsite webserver are you using, if i may ask? Thank you.

alaudet, 2017-02-18:
I have a personal website that I host on a rented server.   I transfer my graphs there and have a web app that I wrote to display the graphs.