alaudet/raspi-sump #10: measurement spikes

morejava, 2016-05-20:
For some reason the measurements spike from a consistent number to 12+ inches... 

Here is a tail -f of my readings although nothing is changing in my sump pump pit.  Ideas?

17:40:02,5.2
17:40:02,5.1
17:40:03,5.1
17:40:04,5.1
17:40:07,5.3
17:40:07,5.3
17:40:10,10.9
17:41:02,8.0
17:41:03,5.2
17:41:03,5.3
17:41:07,5.2
17:41:13,6.2
17:41:13,5.2
17:41:15,5.2
17:42:02,5.2
17:42:02,5.1

alaudet, 2016-05-21:
How often are you taking a reading?  It looks like every second.  Your pi basically does not have time to finish a reading that it is starting another one.

What value have you put in your conf file for reading_interval?

morejava, 2016-05-24:
300

alaudet, 2016-05-24:
That is strange.  If you are taking a reading every 300 seconds (5 minutes) you should not be seeing readings every few seconds like that.  It's like there are multiple processes of rsump.py running.  If you need to take readings only every 5 minutes you should set the reading_interval to 0 and run a scheduled task with cron every 5 minutes. 

Crontab is a much more robust way to schedule activities but is limited to 1 minute as its minimum interval.  The only time you should set reading_interval to anything but 0 is if you need to take more than one reading a minute.  

To schedule a task do the following.

Change reading_interval in your conf file to 0

Run the following command to ensure there are no rsump.py processes running

    sudo killall 09 rsump.py

1 - at console prompt type crontab -e

2 - enter line in crontab as follows;

    */5 * * * * /usr/local/bin/rsump.py

**Note: you may need to run the above command with sudo on Raspbian Wheezy

3 - Save crontab

closed 2016-06-29 by alaudet