alaudet/raspi-sump #12: Question: Does this setup work for a 2.4m deep tank?

ganar, 2016-10-13:
Hello Al,

Do you think that the ultrasonic sensor [of your project](https://www.linuxnorth.org/raspi-sump/) will work on a tank about 2.4 m deep? the documentation on the [Hc-sr04](https://www.amazon.com/Ultrasonic-Distance-Measuring-Compatible-Duemilanove/dp/B00SXZWMCS/ref=as_li_ss_tl?ie=UTF8&qid=1472623587&sr=8-1&keywords=hc-sr04&linkCode=sl1&tag=intorobo-20&linkId=d66cae427e42c34818fbbd9fcfd4a940) is non conclusive; in some pages is only good to 50cm in others is about 3m.

alaudet, 2016-10-13:
The hcsr04 manufacturers manual explains this quite well.

https://www.linuxnorth.org/raspi-sump/HC-SR04Users_Manual.pdf

2.4m is just under 8 feet which should work as the specification says up to 13 feet.  Other factors can affect your readings though such as obstacles.  The only way is to test it and make sure that you have a good line of site from the sensor to the bottom of the tank.  All this to say...it should work, but you need to try and see.

ganar, 2016-10-14:
Thanks a lot for your help @alaudet, and for pointing me to the manual

alaudet, 2016-10-14:
It's my pleasure.

closed 2016-10-14 by alaudet