alaudet/raspi-sump #14: Senor reading consitency?

moontz, 2016-10-14:
Hi - First let me say a huge thanks for posting everything you have done for this.  It has been extremely helpful.  

One question I had (and have been digging more into myself) is how to get the charts to represent a more consistent measurement.   I understand the sensor is not going to give perfect readings and am fine with that.   But I am seeing if I can find a way to interpolate the data in a way that when I am looking at shorter term views of the data, it is not bouncing all over because of the the lower scale for the shorter time period (say 30-90mins).   Seeing if I can somehow avg out the readings to give me a straighter chart reading.  Have you looked into this at all?  if not, no problem.

alaudet, 2016-10-14:
Hi, would you be able to provide me with a few log reading files to have a look?  I can understand that if you only have a handful of readings that your charts may look kind of ugly.  I can certainly give it some thought.

moontz, 2016-10-14:
Sure.  I will email you my test site and you can see the live data directly.   Thanks!

moontz, 2016-10-14:
Actually I forgot my site is down.  But you know what I mean.   When looking at say 60 mins of readings, the charts look a little ugly.   I for example reading every 2 mins I would get, 7.0, 7.4, 6.8, 7.5, 6.8, 6.9, 6.7, 7.5, etc.   Bounces around a bit which makes charts look messy when looking at a smaller timeframe.   Its not a big deal at all as more data comes, they straighten out.   But just was checking if you ever tried to do what I was mentioning above.   If not, no worries.  If I figure something out, I will let you know.

alaudet, 2016-10-14:
I have always accepted that my graphs would have a sawtooth look.  It has always looked fine with anything but the smallest samples.  I am sure you could easily add some code that says something like

if reading < previous reading then reading = previous reading.   that way it would always go up.  You would need to add something to not trigger the code if the reading is more than an inch less (as that would kill any plots that are created when the sump empties. 

A seperate script that normalizes your data before creating the chart would be the way to go.

closed 2016-10-15 by alaudet