alaudet/raspi-sump #17: Not Really a Big Deal

budpi, 2017-01-27:
There seems to be a 

.%f 

at the end of the time stamp for the chart/graph

example:

19:55:00.%f

Any thoughts?

Great Project by the way!!!!

alaudet, 2017-01-27:
Hi this is related to another issue that was closed a while back.
https://github.com/alaudet/raspi-sump/issues/5

There is a bug in matplotlib where when only a few times are plotted it wants to add microseconds to the time.  As you get more readings it will disappear.  Since it was such a low impact issue and the fix would involve building versions of matplotlib not in the repos I have decided to close it.  It's more pain than it is worth.  That being said....and I have not tested this, if you try the latest version of matplotlib (not the preckaged deb with raspbian) it may address it, but I have not bothered to test this as matplotlib is a pain to install at times and I did not want to get into that simply for a small issue like this.

closed 2017-01-27 by alaudet

budpi, 2017-01-28:
Cool. I am not sweating it. Keep up the great work.