alaudet/raspi-sump #18: Pull and Show water_depth

wxflint, 2017-02-22:
Is there a way To show the "water_depth" in TXT on the web page that shows the Graph as well?

alaudet, 2017-02-23:
I am not sure what you mean, you want to see the contents of the csv file in a web page?

wxflint, 2017-02-23:
Hi alaudet, Thanks. That is correct like have the actual reading in text format like below the graph. doesnt have to be all of the readings just that day or hour. Thanks again

alaudet, 2017-02-23:
That's possible.  Just a matter of pulling lines from the csv file and displaying them in the index.html page below the graph.  I can look at it when I have some time, not sure how quickly I could do that though. I will leave this issue open as a feature request.

wxflint, 2017-02-23:
Thanks you, I appreciate your time and service

budpi, 2017-03-02:
A current/last reading would be a great addition to an already great project. Thanks for looking into this.

wxflint, 2017-03-02:
I have been working on it in my spare time. 
https://www.dropbox.com/s/dj2vmxalf49alnw/Photo%20Mar%2001%2C%2010%2054%2022%20PM.jpg?dl=0

budpi, 2017-04-04:
Morning wxflint, 

Can you share a bit on how you made your current measurement viewable on your web page?

Thanks!

gastrog, 2018-02-21:
I'm trying to implement this function too.   @ wxflint  can you update?
@alaudet :: thanks so much for sharing your awesome project!

alaudet, 2018-02-22:
Hey guys, what I have on my own personal website is aside from the actual raspi-sump program.  I just made a quick and dirty on the Pi that you can access using the Lighttpd webserver. 

What I have on my site is the Pi pushing the graphs hourly offsite to my webserver and then some server side code I wrote that parses the graphs into a web page based on the month you want to view.  I would like to release the web side stuff some day but it is setup according to my specific needs and would not be easy as a one size fits all that could be released...and supported.

alaudet, 2022-02-12:
I am adding an overhaul of the web based display on the pi to V2 of Raspisump.  I will still look to accept any contributions before then but it will not be tracked here.

closed 2022-02-12 by alaudet