alaudet/raspi-sump #20: Error Message

sjmorrow26990, 2017-03-12:
Hi
Here is a cut and paste of the error message i'm getting, sorry if it is a stupid error that I caused but the is a great project and my FIRST build on a pi.  The rsump.py file gives me this error when I run it, What can I do to fix it.

Thanks Stan

Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 27, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.4/dist-packages/raspisump/reading.py", line 68, in water_depth
    log.log_reading(water_depth)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/log.py", line 19, in log_reading
    csv_file = open(filename, 'a')
PermissionError: [Errno 13] Permission denied: '/home/pi/raspi-sump/csv/waterlevel-20170311.csv'

alaudet, 2017-03-12:
It's a permissions error on the csv file.  I would need to know what permissions you have on the file.  
Are you running rsump.py with sudo?  You shouldn't have to but if the file was created as root and then you are trying to run it as a regular user then it could be the problem.

What is the output of   ls -al /home/pi/raspi-sump/csv/

This will list all files  in the folder and show the permissions on them.

sjmorrow26990, 2017-03-18:
Hi
can you point in the right direction on how to fix this error.  this is the message i get when i run rsumpchart.py,  

pi@sump:~/raspi-sump $ rsumpchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 24, in <module>
    main()
  File "/usr/local/bin/rsumpchart.py", line 21, in main
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52, in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 736, in loadtxt
    fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv/waterlevel-20170318.csv'

alaudet, 2017-03-18:
Does this file exist on your pi?

'/home/pi/raspi-sump/csv/waterlevel-20170318.csv'

sjmorrow26990, 2017-03-19:
Sorry I should have mentation  that there is no files in the folder, I have tried runny rsump.py, than rsumpchart.py , than resumpwebchart.py.  rsump.py works but the next two give me that error.

thanks

alaudet, 2017-03-19:
Then that is the problem.   The error message is saying exactly that.  The next thing to check is what happens when you try to run the rsump.py program manually.  Are you still getting that permission denied error?

What are the permissions on your csv folder?  Send me the output of 'ls -al /home/pi/raspi-sump/csv'

sjmorrow26990, 2017-03-19:
Thanks for getting back to me....

I chmod the csv folder to 775 and than 777 but still getting the same error.

Here is the outputs

pi@sump:~/raspi-sump $ chmod 775 csv
pi@sump:~ $ rsumpchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 24, in <module>
    main()
  File "/usr/local/bin/rsumpchart.py", line 21, in main
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52                        , in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 736, in loadtxt
    fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv                        /waterlevel-20170319.csv'

pi@sump:~ $ cd raspi-sump
pi@sump:~/raspi-sump $ ls -la
total 48
drwxr-xr-x  9 pi   root 4096 Mar 18 14:21 .
drwxr-xr-x 18 pi   pi   4096 Mar 18 14:44 ..
drwxr-xr-x  3 pi   root 4096 Mar 18 14:42 charts
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 cron
-rw-r--r--  1 root root  132 Mar 18 14:42 crontab
drwxrwxr-x  2 pi   root 4096 Mar 18 14:10 csv
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 docs
drwxr-xr-x  2 pi   root 4096 Mar 18 15:04 logs
-rw-r--r--  1 pi   root 5051 Mar 18 15:07 raspisump.conf
drwxr-xr-x  2 pi   root 4096 Mar 18 14:11 sample_config
drwxr-xr-x  4 pi   root 4096 Mar 18 14:10 web

pi@sump:~/raspi-sump $ chmod 777 csv
pi@sump:~/raspi-sump $ ls -la
total 48
drwxr-xr-x  9 pi   root 4096 Mar 18 14:21 .
drwxr-xr-x 18 pi   pi   4096 Mar 18 14:44 ..
drwxr-xr-x  3 pi   root 4096 Mar 18 14:42 charts
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 cron
-rw-r--r--  1 root root  132 Mar 18 14:42 crontab
drwxrwxrwx  2 pi   root 4096 Mar 18 14:10 csv
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 docs
drwxr-xr-x  2 pi   root 4096 Mar 18 15:04 logs
-rw-r--r--  1 pi   root 5051 Mar 18 15:07 raspisump.conf
drwxr-xr-x  2 pi   root 4096 Mar 18 14:11 sample_config
drwxr-xr-x  4 pi   root 4096 Mar 18 14:10 web

pi@sump:~/raspi-sump $ rsumpchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 24, in <module>
    main()
  File "/usr/local/bin/rsumpchart.py", line 21, in main
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52                        , in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 736, in loadtxt
    fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv                        /waterlevel-20170319.csv'

pi@sump:~/raspi-sump $ cd csv
pi@sump:~/raspi-sump/csv $ ls
README.md
pi@sump:~/raspi-sump/csv $

alaudet, 2017-03-19:
You need to run rsump.py first to create the csv file. Rsumpchart.py is looking for it but it doesn't exist.                                                                                                                                                                                                                                                                                                                                        Al                                                                                                                                                                                                                From: sjmorrow26990Sent: Sunday, March 19, 2017 10:02 AMTo: alaudet/raspi-sumpReply To: alaudet/raspi-sumpCc: Al Audet; CommentSubject: Re: [alaudet/raspi-sump] Error Message (#20)Thanks for getting back to me....
I chmod the csv folder to 775 and than 777 but still getting the same error.
Here is the outputs
pi@sump:/raspi-sump $ chmod 775 csv
pi@sump: $ rsumpchart.py
Traceback (most recent call last):
File "/usr/local/bin/rsumpchart.py", line 24, in 
main()
File "/usr/local/bin/rsumpchart.py", line 21, in main
todaychart.graph(csv_file, filename, bytes2str)
File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52                        , in graph
converters={0: bytes2str}
File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 736, in loadtxt
fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv                        /waterlevel-20170319.csv'
pi@sump:~ $ cd raspi-sump
pi@sump:~/raspi-sump $ ls -la
total 48
drwxr-xr-x  9 pi   root 4096 Mar 18 14:21 .
drwxr-xr-x 18 pi   pi   4096 Mar 18 14:44 ..
drwxr-xr-x  3 pi   root 4096 Mar 18 14:42 charts
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 cron
-rw-r--r--  1 root root  132 Mar 18 14:42 crontab
drwxrwxr-x  2 pi   root 4096 Mar 18 14:10 csv
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 docs
drwxr-xr-x  2 pi   root 4096 Mar 18 15:04 logs
-rw-r--r--  1 pi   root 5051 Mar 18 15:07 raspisump.conf
drwxr-xr-x  2 pi   root 4096 Mar 18 14:11 sample_config
drwxr-xr-x  4 pi   root 4096 Mar 18 14:10 web
pi@sump:/raspi-sump $ chmod 777 csv
pi@sump:/raspi-sump $ ls -la
total 48
drwxr-xr-x  9 pi   root 4096 Mar 18 14:21 .
drwxr-xr-x 18 pi   pi   4096 Mar 18 14:44 ..
drwxr-xr-x  3 pi   root 4096 Mar 18 14:42 charts
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 cron
-rw-r--r--  1 root root  132 Mar 18 14:42 crontab
drwxrwxrwx  2 pi   root 4096 Mar 18 14:10 csv
drwxr-xr-x  2 pi   root 4096 Mar 18 14:10 docs
drwxr-xr-x  2 pi   root 4096 Mar 18 15:04 logs
-rw-r--r--  1 pi   root 5051 Mar 18 15:07 raspisump.conf
drwxr-xr-x  2 pi   root 4096 Mar 18 14:11 sample_config
drwxr-xr-x  4 pi   root 4096 Mar 18 14:10 web
pi@sump:~/raspi-sump $ rsumpchart.py
Traceback (most recent call last):
File "/usr/local/bin/rsumpchart.py", line 24, in 
main()
File "/usr/local/bin/rsumpchart.py", line 21, in main
todaychart.graph(csv_file, filename, bytes2str)
File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52                        , in graph
converters={0: bytes2str}
File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 736, in loadtxt
fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv                        /waterlevel-20170319.csv'
pi@sump:/raspi-sump $ cd csv
pi@sump:/raspi-sump/csv $ ls
README.md
pi@sump:~/raspi-sump/csv $

sjmorrow26990, 2017-03-19:
ok format the sd card and did a fresh install, and I still have the same issue.......

here is all the code output

pi@192.168.1.126's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Mar 19 12:20:33 2017

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@sump:~ $ sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-pip is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@sump:~ $ sudo pip3 install raspisump
Downloading/unpacking raspisump
  Downloading raspisump-1.1.0.tar.gz (69kB): 69kB downloaded
  Running setup.py (path:/tmp/pip-build-og4qe65x/raspisump/setup.py) egg_info for package raspisump

Downloading/unpacking hcsr04sensor (from raspisump)
  Downloading hcsr04sensor-1.5.1.tar.gz
  Running setup.py (path:/tmp/pip-build-og4qe65x/hcsr04sensor/setup.py) egg_info for package hcsr04sensor

Requirement already satisfied (use --upgrade to upgrade): RPi.GPIO in /usr/lib/python3/dist-packages (from hcsr04sensor->raspisump)
Installing collected packages: raspisump, hcsr04sensor
  Running setup.py install for raspisump
    changing mode of build/scripts-3.4/rsump.py from 644 to 755
    changing mode of build/scripts-3.4/rsumpchart.py from 644 to 755
    changing mode of build/scripts-3.4/rsumpmonitor.py from 644 to 755
    changing mode of build/scripts-3.4/rsumpwebchart.py from 644 to 755

    changing mode of /usr/local/bin/rsump.py to 755
    changing mode of /usr/local/bin/rsumpwebchart.py to 755
    changing mode of /usr/local/bin/rsumpchart.py to 755
    changing mode of /usr/local/bin/rsumpmonitor.py to 755
    chmod: cannot access ‘/home/pi/raspi-sump/raspisump.conf’: No such file or directory
  Running setup.py install for hcsr04sensor
    changing mode of build/scripts-3.4/hcsr04.py from 644 to 755

    changing mode of /usr/local/bin/hcsr04.py to 755
Successfully installed raspisump hcsr04sensor
Cleaning up...
pi@sump:~ $ sudo apt-get install python3-matplotlib
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  fonts-lyx libjs-jquery-ui python-matplotlib-data python3-dateutil
  python3-nose python3-pyparsing python3-tz
Suggested packages:
  libjs-jquery-ui-docs dvipng ghostscript inkscape ipython3
  python-matplotlib-doc python3-cairocffi python3-gi-cairo python3-pyqt4
  python3-scipy python3-sip python3-tornado texlive-extra-utils
  texlive-latex-extra ttf-staypuft python-nose-doc
The following NEW packages will be installed:
  fonts-lyx libjs-jquery-ui python-matplotlib-data python3-dateutil
  python3-matplotlib python3-nose python3-pyparsing python3-tz
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 7,601 kB of archives.
After this operation, 22.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
[... 166 lines cut ...]
total 48
drwxr-xr-x  9 pi   root 4096 Mar 19 12:39 .
drwxr-xr-x 18 pi   pi   4096 Mar 19 12:24 ..
drwxr-xr-x  3 pi   root 4096 Mar 19 12:42 charts
drwxr-xr-x  2 pi   root 4096 Mar 19 12:24 cron
-rw-r--r--  1 root root  138 Mar 19 12:39 crontab
drwxrwxrwx  2 pi   root 4096 Mar 19 12:24 csv
drwxr-xr-x  2 pi   root 4096 Mar 19 12:24 docs
drwxr-xr-x  2 pi   root 4096 Mar 19 12:37 logs
-rw-r--r--  1 pi   root 5032 Mar 19 12:37 raspisump.conf
drwxr-xr-x  2 pi   root 4096 Mar 19 12:32 sample_config
drwxr-xr-x  4 pi   root 4096 Mar 19 12:24 web
pi@sump:~/raspi-sump $ sudo rsump.py
pi@sump:~/raspi-sump $ rsumpchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 24, in <module>
    main()
  File "/usr/local/bin/rsumpchart.py", line 21, in main
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52, in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 736, in loadtxt
    fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv/waterlevel-20170319.csv'
pi@sump:~/raspi-sump $

alaudet, 2017-03-19:
You shouldn't have to run rsump.py with sudo.  When you run rsump.py the sensor takes a reading and logs it to waterlevel-20170319.csv  (which is today's date).  That is the problem we need to troubleshoot.  Don't bother running the rsumpchart.py if there is no csv file in ~/raspi-sump/csv folder.  You will always get FileNotFoundError.

The issue is trying to figure out why it's not taking the readings.  It can be any number of things.

Start from the beginning and lets look at your raspisump.conf file.  Is everything set right in there such as the gpio pins, pit_depth and critical_waterlevel.

alaudet, 2017-08-23:
I am closing this issue as I have not heard back.  Will reopen if needed.

closed 2017-08-23 by alaudet