alaudet/raspi-sump #21: Error running rsumpwebchart.py

sginokj, 2017-03-16:
pi@raspberrypi:~ $ /usr/local/bin/rsumpwebchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpwebchart.py", line 25, in <module>
    main()
  File "/usr/local/bin/rsumpwebchart.py", line 21, in main
    webchart.create_chart(homedir)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/webchart.py", line 36, in create_chart
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 52, in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 848, in loadtxt
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 848, in <listcomp>
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "/usr/local/lib/python3.4/dist-packages/raspisump/todaychart.py", line 43, in bytesconverter
    return strconverter(s)
  File "/usr/lib/python3/dist-packages/matplotlib/dates.py", line 261, in __call__
    return date2num(datetime.datetime(*time.strptime(s, self.fmt)[:6]))
  File "/usr/lib/python3.4/_strptime.py", line 494, in _strptime_time
    tt = _strptime(data_string, format)[0]
  File "/usr/lib/python3.4/_strptime.py", line 337, in _strptime
    (data_string, format))
ValueError: time data '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0014:18:11' does not match format '%H:%M:%S'

alaudet, 2017-03-16:
Are you able to simply create a chart with rsumpchart.py?
Have you ever been able to create a chart?
Do you have the csv file present in /home/pi/raspi-sump/csv?
Can I see the contents of the waterlevel csv file? Look through the whole file to see if there are any lines that look corrupted.

sginokj, 2017-03-17:
Yes, that could the issue. I am able to create the graph and it ran fine for a few hours. Now the script is working but I wonder whether it will stop working at 14:18:11.

pi@raspberrypi:~/raspi-sump/charts $ date
Fri 17 Mar 06:23:30 IST 2017
pi@raspberrypi:~/raspi-sump/charts $ ls -al
total 44
drwxr-xr-x 3 pi root  4096 Mar 17 06:23 .
drwxr-xr-x 9 pi root  4096 Mar  5 21:11 ..
drwxr-xr-x 3 pi pi    4096 Mar  5 21:20 2017
-rw-r--r-- 1 pi root    80 Mar  2 20:08 README.md
-rw-r--r-- 1 pi pi   26220 Mar 17 06:22 today.png
pi@raspberrypi:~/raspi-sump/charts $ 

pi@raspberrypi:~/raspi-sump $ taif -f csv/
README.md                waterlevel-20170316.csv  
waterlevel-20170305.csv  waterlevel-20170317.csv  
pi@raspberrypi:~/raspi-sump $ taif -f csv/waterlevel-20170317.csv 
-bash: taif: command not found
pi@raspberrypi:~/raspi-sump $ tail -f csv/waterlevel-20170317.csv 
06:16:05,95.4
06:17:05,95.3
06:18:05,94.8
06:19:05,94.7
06:20:05,95.0
06:21:05,94.9
06:22:05,94.4
06:23:05,94.5
06:24:05,94.3
06:25:05,94.0

alaudet, 2017-03-17:
The time it happened is completely random and shouldn't happen again at that precise time.  I have seen this happen a couple of times over the  past two years.  If it happens more frequently it is not an in issue with the graphing but something happening when the log reading write to file happens.  I am not sure what causes it and will investigate further.

alaudet, 2018-01-30:
Unable to reproduce.  Happens so rarely I am not going to chase it anymore.  If it pops up again I will reopen.

alaudet, 2018-01-30:
issue closed

closed 2018-01-30 by alaudet