alaudet/raspi-sump #22: generating wrong graphs

sginokj, 2017-03-24:
Hi Al Audet,
Please see my csv file :

pi@raspberrypi:~/raspi-sump/charts $ tail -f /home/pi/raspi-sump/csv/waterlevel-20170324.csv 
13:07:48,98.7
13:08:52,100.0
13:09:55,101.3
13:10:58,102.0
13:12:02,103.4
13:13:05,104.3
13:14:09,105.6
13:15:12,105.2
13:16:15,103.7
13:17:19,102.7
13:18:22,101.5
13:19:26,101.1

[image]

You can see it doesnt display the correct graph.

My crontab as below 

# m h  dom mon dow   command
#*/1 * * * * sudo /usr/local/bin/rsump.py
*/5 * * * * sudo /usr/local/bin/rsumpmonitor.py
#*/5 * * * * sudo /usr/local/bin/rsumpmonitor.py
#*/30 * * * * /usr/local/bin/rsumpwebchart.py
pi@raspberrypi:~/raspi-sump/charts $ 

I just removed the today.png and tried running the script, still the graph is wrong.

Any ideas?

alaudet, 2017-03-24:
That's very odd. I have no idea with the information you have provided.  I assume you are running the rsumpwebchart.py manually as it is commented out in your cron?

Can you email me your entire waterlevel-20170324.csv file?

sginokj, 2017-03-24:
Emailed the csv file.

alaudet, 2018-01-30:
I cannot recreate this issue.  It is the only time this problem has been reported.  I will close for now unless i get more info.  Can be reopened later.  Could be something with the local configuration of matplotlib.

closed 2018-01-30 by alaudet