alaudet/raspi-sump #23: raspisump installs improperly with PIP9

alaudet, 2017-08-23:
This appears to be caused by pip 9 using wheels to install and I used sdist when packaging raspisump.  

Older versions of pip still work fine but pip 9.x is the default on Raspbian Stretch that was just released.  if you install raspisump and it does not create the /home/pi/raspi-sump directory you can do the following until I figure this out.

sudo pip3 uninstall raspisump
sudo pip3 install --no-binary :all: raspisump

This will force pip to run all commands in setup.py.

This is temporary as I need to read up more on wheels

alaudet, 2018-01-15:
Still on my radar, sorry to anyone waiting for a fix.  I just haven't had much time the past few months.  New Pi3 coming that I will install Stretch on to test stuff.

lecrote, 2018-03-25:
Good afternoon  Al,

I'm not a newbie in linux but not far from that ;-)
I tried to install Raspisump without succes
I tried these command to overcome the PIP 9 issue that you mention here 
sudo pip3 uninstall raspisump
sudo pip3 install --no-binary :all: raspisump
And when I search in the filesystem there is no raspisump.py anywhere.

Any help would be much appreciate.

Regards

Dan

alaudet, 2018-03-25:
The file is rsump.py and should be located in /usr/local/bin

alaudet, 2019-04-23:
I am closing this issue.  I don't  think there is a solution.   Everything I read seems to tell me that the ugly pip command is what is needed to install extra files.    I am considering a major refactoring of the code to make it more compliant with standards for file locations and releasing a deb for it so it can be installed from the repos.

closed 2019-04-23 by alaudet