alaudet/raspi-sump #24: Time of Day is formatted incorrectly

EcurbNosneb, 2018-01-13:
My time of day comes out, for example, "01-01 04" instead of "04:00:00" -- Is not a huge issue as only the hour is significant. 
[waterlevel-20180113.zip](https://github.com/alaudet/raspi-sump/files/1628826/waterlevel-20180113.zip)

[image]

alaudet, 2018-01-14:
What version of Raspbian are you running.  Did you install matplotlib and numpy from the default repositories or via pip?

EcurbNosneb, 2018-01-14:
Raspbian 9 (stretch)
Python 3.5.3
python-numpy/stable,now 1:1.12.1-3 armhf [installed,automatic]
python-matplotlib/stable 2.0.0+dfsg1-2 armhf

I don't recall installing numpy or matplotlib, but I would have followed
your instructions and hence used apt-get.  I retried each apt-get and pip3
and they all came back as being the most recent (already satisfied).

alaudet, 2018-01-15:
Ok, it's probably something with those two packages as they are later than the versions that are on Jessie.  I ordered a Pi3 this week and it will install Stretch on it to see if I can reproduce the problem.  I have a Pi3 running Jessie right now to test but I also use it as a small NAS and don't want to blow it away.  When I get the new Pi I am going to have to put in a bit of work.

alaudet, 2018-01-17:
Got my new Pi3 today and loaded Stretch.  I replicated your issue.  Now to figure out whats going on.

alaudet, 2018-01-17:
Possible temporary fix.

1) create a file called matplotlibrc in the /home/pi/.config/matplotlib directory.
2)enter the following line in the file called
              date.autoformatter.hour   : %H:%M:%S

Save the file.  Run your chart again.

This formats the date correctly.  I will try and fix in the code itself so that it formats correctly regardless of which version of matplotlib it is using.

EcurbNosneb, 2018-01-17:
That worked.  Looks a lot better. Thanks Bruce

alaudet, 2018-01-17:
Great.  I am going to add a fix to todaychart.py to check for the matplotlib version. I will let you know when it is done.  By adding the setting in matplotlibrc it changes the behaviour user wide.  The fix will only change it for raspisump.  I got it working I think, but I need to test everything and clean it up before releasing the update for download.

cheers :-)

alaudet, 2018-01-27:
Version 1.1.2 released to fix this issue.

sudo pip3 install -U raspisump

Thanks Bruce for reporting it.

closed 2018-01-27 by alaudet