alaudet/raspi-sump #25: Generate charts based on old data

github-t, 2018-02-22:
I haven't been generating charts for awhile but I do have all of my old .csv's.  Is there a way to generate charts based on all of the old .csv's that I have?

alaudet, 2018-02-22:
Hi, it's possible to do, my issue right now is that I pretty much have no time to work on the application aside from the odd bug fix or security fix should it be required.  It would only require a script to walk through the files in your folder using todaychart.py (which would need a few changes).  I am putting it as a feature request but my winter is looking tight right now, I wish I had more time to work on this project again.

Would be great to have a few contributors that knew Python that could take some small things on.

gottifour, 2018-08-28:
I really wish I had the ability to help out myself! I need to learn some Python basics ASAP. I'm sure somebody using your awesome tool knows how to write in Python..

tonymontgomery, 2018-08-29:
I could help out.
I would just need to think out how to best accomplish this with little user interaction to make it "non-programmer" friendly. 
Any ideas of what you would like or expect as far as a flow?

ie: "I click this button and I see this." Or... "I do this and it emails me this." ...etc

alaudet, 2019-05-02:
Closing issue as it is being rolled into a larger request in issue #39 

https://github.com/alaudet/raspi-sump/issues/39

closed 2019-05-02 by alaudet