alaudet/raspi-sump #26: python3-matplotlib - does not work

spankurmonkey, 2018-04-17:
pi@raspberrypi:~/raspi-sump $ sudo apt-get install python3-matplotlib
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package python3-matplotlib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python3-matplotlib' has no installation candidate

alaudet, 2018-04-17:
What Raspbian version are you running?

spankurmonkey, 2018-04-17:
when I try to run it I get this:  
pi@raspberrypi:~/raspi-sump/logs $ rsump.py &
[1] 1058
pi@raspberrypi:~/raspi-sump/logs $ Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 27, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.5/dist-packages/raspisump/reading.py", line 70, in water_depth
    log.log_reading(water_depth)
  File "/usr/local/lib/python3.5/dist-packages/raspisump/log.py", line 19, in log_reading
    csv_file = open(filename, 'a')
PermissionError: [Errno 13] Permission denied: '/home/pi/raspi-sump/csv/waterlevel-20180417.csv'

alaudet, 2018-04-19:
Sorry I have been very busy in my personal life lately and have not had much chance to look at anything. 
The error is saying that there is a permission error in the csv folder.  Are there any files at all in that folder?

Do you even have a csv folder?

ls -al /home/pi/raspi-sump     
ls -al /home/pi/raspi-sump/csv

I would like to see the contents of these folders, If you can run these two commands that would be helpful.

spankurmonkey, 2018-04-19:
Thanks for helping.  Here is the results of those two queries.
pi@raspberrypi:~ $ ls -al /home/pi/raspi-sump
total 44
drwxr-xr-x  9 pi root 4096 Apr 17 22:26 .
drwxr-xr-x 18 pi pi   4096 Apr 17 22:17 ..
drwxr-xr-x  2 pi root 4096 Apr 17 18:37 charts
drwxr-xr-x  2 pi root 4096 Apr 17 18:37 cron
drwxr-xr-x  2 pi root 4096 Apr 19 00:01 csv
drwxr-xr-x  2 pi root 4096 Apr 17 18:37 docs
drwxr-xr-x  2 pi root 4096 Apr 17 20:32 logs
-rw-r--r--  1 pi root 4955 Apr 17 21:31 raspisump.conf
drwxr-xr-x  2 pi root 4096 Apr 17 18:38 sample_config
drwxr-xr-x  4 pi root 4096 Apr 17 18:37 web
pi@raspberrypi:~ $ ls -al /home/pi/raspi-sump/csv
total 44
drwxr-xr-x 2 pi   root  4096 Apr 19 00:01 .
drwxr-xr-x 9 pi   root  4096 Apr 17 22:26 ..
-rw-r--r-- 1 pi   root    80 Jan 23 00:19 README.md
-rw-r--r-- 1 root root   286 Apr 17 22:26 waterlevel-20180417.csv
-rw-r--r-- 1 pi   pi   14976 Apr 18 23:59 waterlevel-20180418.csv
-rw-r--r-- 1 pi   pi    8398 Apr 19 13:29 waterlevel-20180419.csv
pi@raspberrypi:~ $

spankurmonkey, 2018-04-19:
I did open up the csv for today and this is the output in ssh.
00:01:05,7.3
00:02:05,7.5
00:03:05,7.5
00:04:05,7.5
00:06:05,7.5
00:07:05,7.5
00:08:05,7.3
00:09:05,7.3
00:11:05,7.4
00:12:05,7.4
00:13:05,7.4
00:14:05,7.6
00:16:06,7.4
00:17:05,7.6
00:18:05,7.6
00:19:05,7.6
00:21:05,7.8
00:22:05,7.8
00:23:05,7.8
00:24:05,7.6
00:26:05,7.8
00:27:05,7.8
00:28:05,7.5
00:29:06,7.6
00:31:05,7.6
00:32:06,7.5
00:33:05,7.6
00:34:05,7.6
00:36:05,7.5
00:37:05,7.5
00:38:05,7.5
00:39:05,7.5
00:41:05,7.6
00:42:05,7.7
00:43:05,7.6
00:44:05,7.7
00:46:05,7.6
00:47:05,7.9

But when I go to the IP for it the browser does not display a chart.  it only shows a broken image link.

spankurmonkey, 2018-04-19:
The charts directory is empty.  only has readme.md

spankurmonkey, 2018-04-19:
I reran these and it seems matplotlib and numpy is installed but get the above error

pi@raspberrypi:~ $ sudo apt-get install python3-matplotlib
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-matplotlib is already the newest version (2.0.0+dfsg1-2).
The following packages were automatically installed and are no longer required:
  librtimulib-dev librtimulib-utils librtimulib7 python-rtimulib
  python-sense-hat python3-rtimulib
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ sudo apt-get install python3-numpy
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-numpy is already the newest version (1:1.12.1-3).
The following packages were automatically installed and are no longer required:
  librtimulib-dev librtimulib-utils librtimulib7 python-rtimulib
  python-sense-hat python3-rtimulib
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $

tonymontgomery, 2018-04-19:
As a sanity check... can you do the following and post results?

$ python3
>>> import numpy as np
>>> print (np.__version__)
>>> print (np.__path__)

>>> import matplotlib as mat
>>> print (mat.__version__)
>>> print (mat.__path__)

------------------
For example, here are my results:
```
$ python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy as np
>>> print (np.__version__)
1.12.1
>>> print (np.__path__)
['/usr/lib/python3/dist-packages/numpy']
>>> 
>>> import matplotlib as mat
>>> print (mat.__version__)
2.0.0
>>> print (mat.__path__)
['/usr/lib/python3/dist-packages/matplotlib']
>>> 
```

spankurmonkey, 2018-04-19:
pi@raspberrypi:~ $ python -V
Python 2.7.13

pi@raspberrypi:~ $ python -c "import numpy; print(numpy.version.version)"
1.12.1

I don't know the correct commands to get the other stuff.

tonymontgomery, 2018-04-19:
Cool... so there is one difference with matplotlib version.
Please also post the results for "python" (instead of 'python3').

spankurmonkey, 2018-04-19:
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>>
>>> import numpy as np
>>> print (np.__version__)
1.12.1
>>> print (np.__path__)
['/usr/lib/python2.7/dist-packages/numpy']
>>> import matplotlib as mat
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named matplotlib
>>>

spankurmonkey, 2018-04-19:
thanks - did all those step but the last errors out.

pi@raspberrypi:~ $ rm -rf /home/pi/.config/matplotlib
pi@raspberrypi:~ $ rm -rf /home/pi/.cache/matplotlib
pi@raspberrypi:~ $ sudo rm -rf /usr/local/lib/python3.5/dist-packages/matplotlibpi@raspberrypi:~ $ sudo apt-get install python3-matplotlib
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-matplotlib is already the newest version (2.0.0+dfsg1-2).
The following packages were automatically installed and are no longer required:
  librtimulib-dev librtimulib-utils librtimulib7 python-rtimulib
  python-sense-hat python3-rtimulib
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ dir
Desktop    Downloads  Pictures  python_games  Templates
Documents  Music      Public    raspi-sump    Videos
pi@raspberrypi:~ $ cd raspi-sump
pi@raspberrypi:~/raspi-sump $ sudo python setup.py install
python: can't open file 'setup.py': [Errno 2] No such file or directory
pi@raspberrypi:~/raspi-sump $

spankurmonkey, 2018-04-19:
Ok. I reran the chart rsumpchart.py and rebooted and now the chart shows.   the time is off but I think I fixed it so starting tomorrow I believe I will have accurate times with the reading levels.

spankurmonkey, 2018-04-19:
the only thing that is throwing me off is the chart numbers. 
[image]
It shows from 2 to 9 inches.   My sensor is 21" above the bottom of the pit.   so does this mean it's reading a fluctuation of 2 to 9 inches of water in the sump?

# Unit of measure.  You may choose to use metric or imperial measurements.
# The unit variable must be set as metric or imperial and is case sensitive.
# Any other setting will cause an error.
unit = imperial

# critical_water_level is the water level in the sump pit at which point an
# email alert should be triggered.
# If unit = metric then critical_water_level = centimeters
# If unit = imperial then critical_water_level = inches
critical_water_level = 12

# The pit_depth is the distance between the sensor and the bottom of the
# sump pit.
# Like critical_water_level, enter centimeters for metric and inches for
# imperial
pit_depth = 21

# rsump.py should be run with a cron job.  However cron is limited to
# 1 minute as its minimum interval. If you need to monitor the sump level
# more frequently you can set reading_interval to the desired time in seconds.
# Default is 0 meaning only one reading will occur and the program will exit.
# If set to 0 call the script with a cron job and select your interval there.
# If setting to a value other than 0 you should run rsumpmonitor.py from a cron
# job at a regularly interval to monitor the health of the rsump.py process.
# rsumpmonitor.py is part of the raspisump software.  See documentation for
# details.
reading_interval = 0

# The temperature variable will adjust the speed of sound to reflect
# the normal temperature of your basement.
# If unit = metric then enter the temperature in Degrees Celcius.
# If unit = imperial then enter the temperature in Degrees Fahrenheit.
temperature = 66

# Raspi-Sump can also be used in situations where you would want to check if
# the water_level is too low. Sump pits are concerned with high water levels
# but a drinking water cistern would require an alert when it is low.  For that
# reason you can set the following value to 'low'.  Leave this value as is unless
# you specifically want alerts when the water drops below a certain level.
alert_when = high

tonymontgomery, 2018-04-19:
Good deal - it's working.
In your photo, I would ignore the readings at "2". You are most likely getting reading anomalies.
This could all be affected by sensor position, how much space for sensor to take reading, etc.

If you look at it, you know it didn't go from 2 inches back up to over "8" that quick.
I would say water level gets to around 8.5 inches and pump is kicking on... drains pit to approx 4 inches.

You may have to play with the position of sensor. Wait a day, check graph, move if needed, etc.
But overall I think it looks good. I have never got mine to look as clean as the author's graph. But I can tell it's working and that's all I care about.

Your real reading is something like this:

[image]

spankurmonkey, 2018-04-19:
Thanks for your help.... with the time correction it seems to be working as expected.  I just have the sensor taped over a hole that allows the power line for the pump to exit so not sure if it has a clear line of sight to the bottom.  I am not sure how wide the sensor reads so like you said may have to position it around.

I wish I could have gotten this installed without issue... not sure what I did wrong.

alaudet, 2018-04-20:
Thank you tonymontgomery for stepping in and answering.  For the chart, ya it really looks like interference on the sensor.  But usually it will show a higher waterlevel when it picks up the floating ball or something else.  This can be mitigated by sinking a tube into the pit and placing your sensor over the tube which focuses the sonar down and doesn't pick up all kinds of other garbage in the pit.

Glad you got it doing and sorry I could not spend more time troubleshooting with you.

Is everything now resolved?

spankurmonkey, 2018-04-20:
Al - yes it seems to be working fine..... can you tell me more about the tube your speaking about?
[image]

spankurmonkey, 2018-04-20:
So today I moved the sensor - now it has a direct line of sight to the pit and no obstructions.   Indoing so the sensor is now 18.5" from the bottom of the pit.... can I use 18.5 in the config or must it be a whole number?

alaudet, 2018-04-20:
Needs to be an integer, so no decimals.

spankurmonkey, 2018-04-20:
I changed it back to a integer -  question... if I use the tube will the artifact go away... here is my latest chart and this is with the sensor point down.. directly to water with no obstructions in the way.
[image]

alaudet, 2018-04-20:
Not sure what you mean by the artifact.  You should get a more sane line by cutting out interference.  But you will not really know until you try.  Just setup something temporary with any cylinder on hand to test.  I have never actually done this but have had other users who reported success.

closed 2018-09-23 by alaudet