alaudet/raspi-sump #29: NEwbe Question

rescuemoe, 2019-01-01:
Hello,
First I would like to say thank you for this program.
Regarding email address
Does Gmail still work was having issues with authentication errors.
What email services do work the best for setting up the email messages.
And is there any free services for sending text SMS messages
thanks!!!!
Jeff

alaudet, 2019-01-02:
Hi, Gmail uses Oauth2 for authentication.  For myself I created a throwaway gmail account and relaxed the security setting to not use Oauth2. 

Oauth2 is kind of overkill in my opinion.

I explain how to do it in issue #6 at this link

https://github.com/alaudet/raspi-sump/issues/6

Cheers

rescuemoe, 2019-01-02:
Happy new year!
Thank you thank you I overlooked a typo in the config file and re-reviewed all my setting. Was not sure if it was me or gmail but since I know you still currently use Gmail it had to be me.
Got everything working
Is there a way or an open request for some sort of heart beat email ie: 1 email a day so you know everything is still running an on-line?
Thank you again this is so awesome!

alaudet, 2019-01-02:
There is no open request but if you submit a feature request detailing what you would like I can look at it when I have time.  Although I have to admit I have not worked on RaspiSump in some time.  It's been stable for me and other things in life have taken precedence.  I would like to get back into the program and add features I have wanted previously.  It's just a matter of committing to it again.  That being said if something breaks due to a dependency in a library  I am using I would try to fix it asap.

closed 2019-01-02 by alaudet

rescuemoe, 2019-01-04:
One other question, I have 3 sump pumps in my basement. Little over kill by the previous home owners. Wanted to do this on three . Is there a way to hook up 2 or even 3 sensors to one pie, instead of going the route of have separate pie's for each pump.
Thanks again and Happy New Year

alaudet, 2019-01-04:
Its technically possible but not with this program.