alaudet/raspi-sump #30: different untrasonic sensor

rfeng74, 2019-01-13:
Hello, I sucessfully built and am using a , kudos to Al Audet and the community for an awesome system!

I found an alternate sensor which looks to be virtually identical in terms of performance and capability, however its waterproof and appears to have the sensor remotely connected via cable.  The sensor is: JSN-B02 (SP0433B02) Waterproof Ultrasonic Ranging Module.  Here is a link:  https://www.elecrow.com/water-proof-integrated-ultrasonic-ranging-module-jsnb02-p-1202.html

Would this sensor work with raspi-sump system as is?  If so this might be easier to mount inside a damp environment, better reliability and remotely locate the PCB for the sensor with the Raspberry Pi.

alaudet, 2019-01-14:
Hi, I haven't tested the JSN-B02 but I have tested the JSN-SR04T which looks very similar to this one and it works fine.  It has both Trig and Echo pins which is what is needed by the hcsr04sensor module in raspisump.  I am pretty sure it would work fine, the only difference with the HCSR04 may be in the distances and times between sound pulses.  

The only thing I would check would be the maximum and minimum ranges to get readings, but I am almost certain this sensor would be fine.  If it works please let me know and I will add to the documentation that it works.

thanks

rfeng74, 2019-01-16:
I'll see if I can find datasheets to compare the characteristics, specs, If the JSN-B02 is similar and seems workable, I'll purchase one to test out.  I'll report back once I have the facts.

rfeng74, 2019-03-01:
The only sensor I could find with a remote transducer was the JSN-SR04T 2.0 not hte JSN-B02.

This is the one I purchased:  https://www.amazon.com/Waterproof-Ultrasonic-JSN-SR04T-Integrated-Transducer/dp/B07FQCNXPP/ref=sr_1_3?keywords=JSN-SR04T&qid=1551412667&s=gateway&sr=8-3

The JSN-SR04T 2.0 remote transducer has a ~ 7.5 foot or 2.3 meter wire (cable) to the PCB board which connects to the Raspberry Pi I/O pins (which is great because you can relocate all water sensitive components far away in a dry place).  The only drawback to the JSN-SR04T 2.0 is that its minimum accurate distance is about 20 cm or 8 in whereas the HC-SR04 has a minimum accuracy distance of ~ 2cm.  The reason for the difference in the sensor range is 1 vs 2 sensors, the HC-SR04 has a separate transmitter and receiver so more ultrasonic wave detection can occur sooner thus a shorter accuracy range.  The JSN-SR04T 2.0 has a single transceiver (functions both as a transmitter and receiver).  The JSN-SR04T 2.0 needs to wait longer for accurate receiver detection by allowing the transmitter 'ringing effect' to dampen (fade away).  The drawback of the HC-SR04 is its not waterproof (I already ruined one when water ran across the floor to the sump pit).

I've had the JSN-SR04T 2.0 running for a solid month and it's been stable.  I am getting similar readings compared to the HC-SR04 (4 inches low to 16 inches high).  Another good feature of the JSN-SR04T 2.0 is it can be easily mounted in a drilled out hole on hte sump cover or fits perfectly in a 1/2 inch PVC conduit clamp (see attached pic below for my test setup).

I started looking for watertight enclosures for the HC-SR04 but none exist, apparently any medium other than air in front of the transmitter or receiver sensors causes measurement issues.  There are semi-protective mounting brackets exist for the HC-SR04 (just search for "hc-sr04 bracket") and there are also 3D printed cases that provide more protection. (https://www.yeggi.com/q/ultrasonic+sensor+case/).

Finally, I did see this other sensor which is waterproof and has a separate transmitter and receiver.  Specs say minimum distance is reported at ~ 3cm.  (https://github.com/ritesht93/waterproof_small_blind_ultrasonic_sensor_DYP-A02YYxx_v1.0)

Long term, I may stick with the JSN-SR04T 2.0, it seems to be providing a consistent accurate top water level threshold for me at ~ 16 inches (for the past 30 days of readings), the HC-SR04 was also consistent at 16" top water level readings for 30 days.  My total pit depth from the sensor is set in raspi-sump at 27 inches so this means the sensor is measuring about a 11 inch air gap before the sump pump activates to pump out the water from the sump pit.  If I had a ~8 inch or less air gap from the sensor to the top water level, I would likely switch to the HC-SR04 or another short range sensor (2-3 cm minimum range).  I think the key to the shorter range accuracy is likely the separate transmitter and receiver units in the ultrasonic probe.

[image]

alaudet, 2019-03-02:
Great stuff, I have the JSN-SR04T but have never put it through its paces like you have.  Good to see that it is a solid alternative.  I really like the way you mounted that.  Do you mind if I use your picture on my Raspi-Sump web page, would be good to show this to those who need a water resistant sensor as opposed to the HC-SR04

rfeng74, 2019-03-02:
Yes, please feel free to use anything.  Note that the setup I have is temporary.  I have a brick holding the board where the sensor is attached.  If wasn't going to place a cover over my sump pit I would attach the sensor in a similar fashion but to the side of the underlying larger board in the picture.

However in my case I plan to place a full plexiglass cover over my pit (because I have a radon system) and will cut a hole in the plexiglass cover for the sensor and likely use a rubber grommet around the sensor so if I need to remove or replace the sensor, it's easy to do.  The rubber grommet will provide an airtight seal for the radon system.  I should have the permanent setup complete an a few months.  A few months because I'm having a contractor remove my current sump pit and replace with a much larger sump pit so I can place 3 pumps (2 primary, 1 battery backup).  Once I get my permanent setup, I will give you pictures, details on that as well.

alaudet, 2019-04-14:
Thanks for the example.  I have posted a picture of your waterproof sensor on my website to show example usage as compared to the HCSR04.

Cheers

closed 2019-04-14 by alaudet