alaudet/raspi-sump #31: color changes for charts can't find where to change y-axis values color (in, cm)

rfeng74, 2019-02-26:
I've modified todaychart.py to change the chart colors (title, water level plot line, gridlines, x and y-axis labels, x-axis values.  I have inches for the y-axis values and have tried various places in todaychart.py code but cannot find where to change the y-axis value color (4, 6, 8, 10, 12, 14, 16 inches).  Any help where to change color for the y-axis values? (I found where to change colors for everything else).

I'm a complete novice on GitHub but can below is my modified todaychart.py code if that is helpful.  I've changed the line color to green, background to black, gridlines, title, axis labels, x-axis values to white.  I also enlarged the plot area.  I added comments to the end of each line I modifed.

[todaychart.txt](https://github.com/alaudet/raspi-sump/files/2903311/todaychart.txt)

alaudet, 2019-02-26:
Hi, ya that would be a matplotlib feature.  I used that library to create the charts.  Matplotlib is a bit of a bear to work with, if you are like me, only needing it for this one thing.

Your best bet is to reference the matplotlib documentation at https://matplotlib.org/contents.html

I haven't touched matplotlib in a couple of years.  After I got the graph the way I liked it I really haven't needed it.

If you don't have any luck let me know and I can take a look, but I am not expert with it by any stretch.  Matplotlib has  github repo, if you provide your code there someone should be able to answer it fairly easily.

https://github.com/matplotlib/matplotlib

rfeng74, 2019-02-27:
Thank you, will try the documentation and/or the repo group and report back with results.

ba05, 2019-02-27:
This would probably be buried in matplotlib’s y-axis spine minor and major ticks.

rfeng74, 2019-02-27:
alaudet and ba05, thank you for your suggestions.  I tried researching the links you provided vs the existing todaychart.py code but was having trouble correlating the existing code to the documentation.  Then I had an idea....add a line of code to specify (overwrite the default, customize) the y-axis values similar to what I thought was happening for the x-axis values.  It worked!

I added this new line which changed the y-axis values to white.

` plt.yticks(rotation=0, color='#FFFFFF') # inch or centimeter (y-axis) values color`

My next question is proper coding, would it be preferred to have colors stored in the configuration file and have todaychart read those values, assign to variables in todaychart.py?  If so I can attempt to work on that next.  Attached below is my modified todaychart.py which has a larger chart area, black background, green plot line (water depth vs time) and white for chart title, chart last update timestamp, x-y axis labels, x-y axis values.

[todaychart2.txt](https://github.com/alaudet/raspi-sump/files/2912755/todaychart2.txt)

alaudet, 2019-02-28:
Good work, glad to see you got it going to your liking.

If customizing colors is a feature that you want to add in a program then using the config file makes sense.  I just never had the need to do it so I coded it statically.  It's something I would consider if someone forked the code and submitted a working pull request.

I only have one requirement on raspisump version 1 changes to the config file though.  Any change to the config file cannot break the app if someone doesn't add them to their config file.  So that would require checking for the value in the config file and if it is not there using defaults in the code.  If version 2 was ever released I would refactor the code as there would be a requirement to redo your config file.

If someone updates to the latest version1 I don't want to overwrite existing config files.  But for a version 2 I think it would be fair game for users to expect major changes and start with a fresh config file.

The limited times I did that I called the new value in the config file and when it wasn't there it caught an error and then issues the default.

For example, check the file alert.py.  I wanted to add a time interval in the config file.  So try to fetch the alert_interval from raspisump.conf.  Woops there is none, throw an exception error and just assign it a 5...so every 5 minutes.  That way someone upgrading to the latest version would  not have to touch their config file, the old one would work.

```
# If item in raspisump.conf add to configs dict above.  If not then provide
# a default value
try:
    configs['alert_interval'] = config.getint('email', 'alert_interval')
except configparser.NoOptionError:
    configs['alert_interval'] = 5
```

Hope that makes sense.

rfeng74, 2019-02-28:
Yes that makes perfect sense.  Thank you.

closed 2019-02-28 by rfeng74