alaudet/raspi-sump #32: Key Error - email - Key Error pit

spankurmonkey, 2019-03-13:
I've tried installing and running rsump.py and rsumpchart.py and I get these errors:
```
pi@raspisump:~ $ rsump.py
Traceback (most recent call last):
  File "/usr/lib/python3.5/configparser.py", line 1135, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'email'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 16, in <module>
    from raspisump import reading
  File "/usr/local/lib/python3.5/dist-packages/raspisump/reading.py", line 15, in <module>
    from raspisump import log, alerts
  File "/usr/local/lib/python3.5/dist-packages/raspisump/alerts.py", line 27, in <module>
    configs = {'email_to': config.get('email', 'email_to'),
  File "/usr/lib/python3.5/configparser.py", line 778, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.5/configparser.py", line 1138, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'email'
pi@raspisump:~ $ rsumpchart.py
Traceback (most recent call last):
  File "/usr/lib/python3.5/configparser.py", line 1135, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'pit'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 11, in <module>
    from raspisump import todaychart
  File "/usr/local/lib/python3.5/dist-packages/raspisump/todaychart.py", line 25, in <module>
    configs = {'unit': config.get('pit', 'unit')}
  File "/usr/lib/python3.5/configparser.py", line 778, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.5/configparser.py", line 1138, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'pit'
pi@raspisump:~ $
```

alaudet, 2019-03-13:
That error says  that when looking for the sections email and pit on raspisump.conf it is not finding those sections.

Do you have a raspisump.conf file in /home/pi/raspi-sump?

spankurmonkey, 2019-03-13:
@alaudet 
I just moved it there and then tried rsump.py
```
pi@raspisump:~/raspi-sump $ dir
bin  changelog  conf  cron  docs  LICENSE  MANIFEST.in  raspisump  raspisump.conf  README.md  setup.py  tests
pi@raspisump:~/raspi-sump $ rsump.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/raspisump/reading.py", line 49, in water_reading
    raw_distance = value.raw_distance(sample_wait=0.3)
  File "/usr/local/lib/python3.5/dist-packages/hcsr04sensor/sensor.py", line 90, in raw_distance
    raise SystemError('Echo pulse was not received')
SystemError: Echo pulse was not received

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 24, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.5/dist-packages/raspisump/reading.py", line 66, in water_depth
    water_depth = water_reading()
  File "/usr/local/lib/python3.5/dist-packages/raspisump/reading.py", line 51, in water_reading
    log.log_errors("**ERROR - Signal not received. Possible cable or sensor problem.")
  File "/usr/local/lib/python3.5/dist-packages/raspisump/log.py", line 46, in log_errors
    error_log = open("/home/pi/raspi-sump/logs/error_log", 'a')
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/logs/error_log'
```

spankurmonkey, 2019-03-13:
Seems now the hcsr04 sensor is not working... I left it in the sump since last year.  I had to recycle the PI that was on it and now I am in fury to get it going again as mother nature is down pouring where I live and the snow pack is melting....  any minute my sump will be wet.

spankurmonkey, 2019-03-13:
@alaudet   Also, after following the install I notice that in the install.md my directories do not match the list in the install.md

```
* raspi-sump/sample_config/raspisump.conf (all configurations for raspisump).
* raspi-sump/csv (location of waterlevel readings to csv file)
* raspi-sump/charts (location of charts if using rsumpchart.py)
* raspi-sump/logs (location of rsumpmonitor.py logs if using raspisump as a continuous process)
* raspi-sump/web (all files needed for the optional pi webserver install)
* raspi-sump/cron (example crontab for scheduling readings)
```

My DIR
`pi@raspisump:~/raspi-sump $ bin  changelog  conf  cron  docs  LICENSE  MANIFEST.in  raspisump  raspisump.conf  README.md  setup.py  tests`

spankurmonkey, 2019-03-14:
@alaudet Should I have followed a different set up?  I am using RPI 2 B v1.1 and I installed Raspbian Lite and then followed the install steps here https://github.com/alaudet/raspi-sump following the Python 3 Recommended for Raspbian Jessie

spankurmonkey, 2019-03-14:
@alaudet   Seems I had some broken wires and ended up replacing those and the sensor.... 

I now get this----
```
pi@raspisump:~/raspi-sump $ rsump.py
Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 24, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.5/dist-packages/raspisump/reading.py", line 70, in water_depth
    log.log_reading(water_depth)
  File "/usr/local/lib/python3.5/dist-packages/raspisump/log.py", line 19, in log_reading
    csv_file = open(filename, 'a')
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv/waterlevel-20190314.csv'
```

spankurmonkey, 2019-03-14:
@alaudet    So I wiped the SD card tried again - Raspbian full - got to these steps...

`sudo pip3 install raspisump`

Check DIR  - no raspi-sump or associated file directory per the install.md

then I tried  sudo pip3 install --no-binary :all: raspisump

Check DIR - no raspi-sump or associated file directory per the install.md
```
pi@sump:~ $ dir
Desktop  Documents  Downloads  Music  Pictures  Public  python_games  Templates  Videos
```

I have no clue what I am doing wrong...

spankurmonkey, 2019-03-14:
@alaudet  so I install git and then do a git clone

I get some of the directories found in install.md but not all of them.

```
pi@sump:~/raspi-sump $ dir
bin  changelog  conf  cron  docs  LICENSE  MANIFEST.in  raspisump  README.md  setup.py  tests

```

I tried running setup.py thinking that would complete the install but got this back

```
pi@sump:~/raspi-sump $ sudo python setup.py
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied
```

spankurmonkey, 2019-03-14:
@alaudet finally got it working - would have to trace back all of my steps but it was not a clean install so I am not sure if I am not following directions correctly or the instructions are outdated?   either way, glad its up an running and always appreciate the use of this little monitor.

spankurmonkey, 2019-03-14:
@alaudet Must have a permissions issue or something - cannot figure it out but if on the web page I try to navigate to past water level readings I get a 403 - Forbidden error.

spankurmonkey, 2019-03-17:
Too bad this part is not made clear in the instructions.

sudo chown -R www-data:www-data /home/pi/raspi-sump/charts/

That one line resolved the 403 errors from the charts dir.

alaudet, 2019-03-17:
Sorry I have not been able to respond. I scanned through the thread and there seems to be a few things going on.  It would be helpful if you opened a seperate issue for seperate problems and not everything under one issue.  

I don't have much time to devote to support right now and a thread with this much stuff in it is a little overwhelming.

Were you able to resolve all of your issues.  If yes let me know and I will close this issue in the tracker.  If not please make sure each different problem you have is opened into a seperate issue.

I am not sure why www-data needs to be the owner.  All that user needs is to read the value and I believe the default values allow that.  I would need to understand what is different from your install to the default.  But since you got it going then all good.

spankurmonkey, 2019-03-17:
@alaudet You can close it is running correctly now.

closed 2019-03-17 by alaudet