alaudet/raspi-sump #33: Add Oauth2 for authentication

alaudet, 2019-03-29:
Gmail requires Oauth2 for authentication. Would like to add to Raspi-Sump so that users don't have to relax their gmail security settings.

rtaft, 2019-07-08:
Why not use Application Passwords?  It works fine for sending emails through gmail, though it does not work for sending SMS through google voice.

alaudet, 2019-07-11:
Thanks I will look into it.  I didn't know the option existed.

alaudet, 2022-02-11:
Removing this requirement.  Use Google App Passwords if you do not want to allow access to less secure apps on your account.

Login to your google account to create the password.

Instructions available at  https://support.google.com/accounts/answer/185833?hl=en

Thanks @rtaft  .   I had forgotten about this.

closed 2022-02-11 by alaudet