alaudet/raspi-sump #34: Statistics on Pump Runs or Gallons emptied

jchuckw, 2019-04-04:
Have you given any thought to adding in statistics of say how many times the pump shows to run in a day, and perhaps with a place to specify gallons removed per pump time, how many gallons removed?

I realize this can be done visually for a day or so, but would be nice to be able to look over time and have statistics on how many times per day, per week, looking at it seasonally, etc.   But the base metric would be using the data to identify how many times the pump ran in a day, and logging it in some way.

alaudet, 2019-04-04:
I have given this tonnes of thought over the past couple of years and have code running (ugly code) on my local pi that calculates the number of times the pump goes off in a day and the amount of gallons pumped out.

However it is challenging as the code is so specific to my own pump and pit.

you have to first identify when the pump goes off from the csv file by viewing patterns.   Then it also depends on your pump and how much water it throws --- which also has dependency on how high it is throwing it.  I actually emptied into a 5 gallon bucket and timed it to get a gallons/minute.

I would be happy to share with you what I have but probably wouldn't be able to do that until the weekend.  I actually wrote a small web app in python/flask that runs on the pi that gives me pretty good stats.

jchuckw, 2019-04-04:
That would be great.  It was the identifying when the pump emptied that I was not sure the easy way to do.  I could grab logs in excel or something and do it, but didnt know how to script it.

I might have been oversimplifying it, but was thinking of the water removed in terms of diameter of pit, high point, and low point - calculating volume, etc.   But would have to account for space taken up with pump, etc - so it would for sure be an estimate.  And depending on specific set up with check valve, some could have some backwash into pit I suppose?.

Anwyay - thanks much!

spankurmonkey, 2019-04-04:
@jchuckw @alaudet 

I have done the following - https://thingspeak.com/channels/737963

This is poorly coded/bastardized python3 script is running on the PI that is running Raspi-Sump - I am not knowledgeable enough to how to integrate it into Al's code.

Then I am using a Thingspeak widget on my android to get the stats... very helpful because we have had flooding in our area and going into the rainy season.   Times in prior years my pump would run almost ever two minutes for days on end.

[image]

tonymontgomery, 2019-04-04:
@spankurmonkey  - very nice!

alaudet, 2019-04-04:
@spankurmonkey   wow....ok I want this  :-)

alaudet, 2019-04-04:
@jchuckw  The way I did it was to estimate how long the pump was emptying but looking at the minutes passed while the water level was decreasing minute by minute.  Then calculate how many gallons get pumped out during that time.

I only have access remotely on the pi with a text based browser but it shows the following.  I will post a screenshot tonight from graphical web browser.

```
Logo

   Today's Water Level

   Waterlevel_2019-04-04_15:33

   File:               waterlevel-20190404.csv
   Number of Readings: 933
   Gallons:            360
   Sump Empties:       8
   5 Gallon Pails:     72
   Minutes per empty:  2.3

```

alaudet, 2019-04-05:
Here is a web look.  I'm not a web designer obviously.  :-)  
Also I can bring up any past csv file to view stats for those days.  I haven't updated the page in some time, as it is not referencing historical data from 2017, 2018, 2019.

[image]

Here is an exceptionally wet time of year in 2016

[image]

jchuckw, 2019-04-08:
I believe I have this working right now.   Just sharing what I ended up with in case you wanted to implement or others wanted to know.   Not the cleanest way, but python is far from my expertise.  I modified the log.py to check the current reading vs the last one, and if a difference of >=2.5 (could be whatever you wanted) then it increments a counter.  My sump empties pretty quickly so I don't have any double entries like you might see with this method.  It is only about 4.5 inches or a little more that it moves down each time it empties.   You could add some logic that looked at the last 2-3 readings to ensure no double counts if needed I suppose.   I also added this number into the web page, so I can easily keep an eye on this to make sure the number counter was working properly.

I then added a script that creates a long term log that is set to run at 1 minute before midnight.  It logs the date, and the number of the daily empties, then resets the counter for the next day.   Also copies the log file over to the charts, so I can easily grab that from web page, and pull it into excel to look at trends, etc when I want to.

alaudet, 2019-04-08:
ok, cool.  I am just looking over the code that i wrote to do the same thing and I am scratching my head at a couple of places.   If you send me an email (email on github page) I can reply to you and send you what I did for another perspective.  But as you have already figured out, there are no two pumps alike so it's hard to release it as it probably would not work for others without some modifications.

alaudet, 2019-04-14:
@spankurmonkey    https://thingspeak.com/channels/737963

I love this.  Very nice

alaudet, 2019-04-23:
I will close this issue for now.  I appreciate the discussion on this topic.  It's made me think of possibilities.  We can reopen if you want to continue.

closed 2019-04-23 by alaudet