alaudet/raspi-sump #35: Temperature Sensor

jchuckw, 2019-04-15:
I had an issue in regards to temperature used.   I have an internal and an external sump.  The internal is fine to use a set temp because it does not fluctuate enough to matter, but the external one can fluctuate up to 100 degrees throughout the year, and even up to 30 or so degrees in any given day.   Just wanted to share how I set it up in case anyone had a similar problem.

I used a DS18B20 sensor (the waterproof version of it).  This was very easy to incorporate into the code to use the temp the sensor gave.  Several places online you can see how to wire this into the pi - it is very straight forward.

Need to turn on 1 wire support - this can be done via the GUI:
preferences / raspberry pi configuration 
[image]

Or can be done via command line   "sudo raspi-config"

Need to install w1thermsensor via pip3
"sudo pip3 install w1thermsensor"

Then we need to modify one file to use this specific reading, and not the temp set in the raspisump.conf file

Have the file attached.  Below is what I changed in case you want to edit it yourself.

Edit  /usr/local/lib/python3.5/dist-packages/hcsr04sensor/sensor.py

At the top, right after the import statements, add the following lines

from w1thermsensor import W1ThermSensor
sensort = W1ThermSensor()
tempc = sensort.get_temperature(W1ThermSensor.DEGREES_C)

In the def __init__ section just below, change the following:
change from:
self.temperature = temperature
to:
self.temperature = tempc

The speed of sound code below this is designed to calculate based on C and not F temp, so it is easier to just calculate it that way directly.  If you need to use imperial (like I did) - then comment out the following

       # if self.unit == 'imperial':
       #     self.temperature = (self.temperature - 32) * 0.5556
       # elif self.unit == 'metric':
       #     pass

       # else:
       #     raise ValueError(
       #         'Wrong Unit Type. Unit Must be imperial or metric')

If you are using Metric, you can leave as is, or comment out, either way is fine (as long as you dont have something other than imperial or metric in the raspisump.conf file)

That is it.  This will then pull a current temp each time you run it instead of using the estimated temp in the raspisump.conf.

As Python is not my main thing, if there is an easier or better way to do it, please feel free to offer any changes - but this appears to be working great on current raspberry pi 3b+ and current raspbian.
[sensor.zip](https://github.com/alaudet/raspi-sump/files/3080873/sensor.zip)

alaudet, 2019-04-15:
This looks great, thanks for sharing.  I often thought about adding it to hcsr04sensor but didn't because I wasn't sure it would make enough difference at short distances.

I know I could do the math, but, on the distances  you are measuring, what is the difference in the readings you are getting when you account for temperature vs not accounting for it.  Those are some pretty big temp swings.

I am going to purchase the sensor to play around with it.  I think  this could be something worth adding.

jchuckw, 2019-04-15:
The base difference in speed of sound over that temp spread is appx 10.7%.  For most sump setups it probably does not make much difference due to the short distance, but my external sump is 12 inch diameter, and 4 feet deep.  So, in theory, that would make it what - up to appx 5 inches off?   More at the lower depths, and less at the higher depths - since it would be traveling shorter distances as the level got higher in the sump.   I could just pick a number in the middle, and it would be at most 2.5-3 inches off at any given time.   I did it partly because I thought it needed done and partly because I am just crazy enough that if I am measuring it, I want it to be as accurate as possible.

alaudet, 2019-04-15:
Great info thanks.  I am going to pick up the sensors and see if I can incorporate it.  I don't think I would add it to hcsr04sensor though.  I think it is something that applications should probably be in control of.  The hcsr04sensor library will take a temp variable that is passed to it.  So the logical place for me to add the code would be right in raspisump.  Let application authors who use the hcsr04sensor library decide it it makes sense to be even caring about the temperature.

dorrsr, 2021-08-17:
Hi,
I'm curious when you say that you would modify raspisump, are you talking about adjusting the water_reading() function so it pulls the temperature right before it calls for the sensor measurement?

def water_reading():
    """Initiate a water level reading."""
    pit_depth = configs["pit_depth"]
    trig_pin = configs["trig_pin"]
    echo_pin = configs["echo_pin"]
    **temperature = configs["temperature"]**  <-- change line to read from the temp sensor
    unit = configs["unit"]

    value = sensor.Measurement(trig_pin, echo_pin, temperature, unit)

I ask because I haven't figured out the best place to change it yet.

alaudet, 2021-08-17:
Hi, yes that would be the place to change it.  Suppose you had a function to call your temperature().  That function would have code that runs each time rsump.py is initiated and return  your value to include it in the sensor.Measurement class. 

So instead of  temperature = configs["temperature"] (which comes from raspisump.conf and is static)   you would have temperature = temperature()

The function it calls would be something like;

```
def temperature():
    temperature = _do something_
    return temperature
```

alaudet, 2022-02-12:
I have added this request to the Version 2 feature request list.

closed 2022-02-12 by alaudet