alaudet/raspi-sump #36: Error - rsumpchart.py command

jlackey84, 2019-04-21:
Setting this up, I executed the rsump.py command with no apparent issues.
When I run the rsumpchart.py command I get the following:
$ rsumpchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 24, in <module>
    main()
  File "/usr/local/bin/rsumpchart.py", line 21, in main
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.5/dist-packages/raspisump/todaychart.py", line 54, in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 860, in loadtxt
    fh = iter(open(fname))
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/raspi-sump/csv/waterlevel-20190421.csv'

Worth noting that there is a csv folder, but no .csv file within (only a readme)

Thanks in advance!

alaudet, 2019-04-21:
When you installed raspisump did you use the following command?

**sudo pip3 install --no-binary :all: raspisump**

It has to do with the changes made to the pip installer.  /home/pi/raspi-sump should have the following folders after initial install;  (if not use the command above to install) 

raspi-sump/sample_config/raspisump.conf (all configurations for raspisump).
raspi-sump/csv (location of waterlevel readings to csv file)
raspi-sump/charts (location of charts if using rsumpchart.py)
raspi-sump/logs (location of rsumpmonitor.py logs if using raspisump as a continuous process)
raspi-sump/web (all files needed for the optional pi webserver install)
raspi-sump/cron (example crontab for scheduling readings)

jlackey84, 2019-04-21:
Thanks for the fast reply!  I did use that command; I tried the original command first but the directory did not create.  The directory now matches what you posted, including the csv folder, but it looks like csv files aren’t being created.

alaudet, 2019-04-22:
what happens when you run /usr/local/bin/rsump.py from the command line?  Do you get an error?

It's hard to tell with limited info.  Has it ever created the file properly?  is the conf file configured properly?  I just don't have enough to go on.

jlackey84, 2019-04-22:
I understand, I’ll try to provide as much info as possible (bear in mind I’m pretty new at this).  I have run the command, not much happens.  No errors or any issues that I can tell.  It’s never created the csv file.  I’m fairly sure the conf file is configured properly

jchuckw, 2019-04-22:
I had that or at least a similar issue initially when I did not use the longer install command - then went back and used it.   I eventually removed and then re-installed to make it work properly.  If all else fails, you could do that.

I suggest if you want to try before that to make sure that the permissions on that folder did not get messed up.   If you open up a terminal, you can do:
cd raspi-sump
Then type in:
ls -la
you should show that the owner and group of the folder is "pi"  If it shows root, or something else that could be an easily fixable problem.

jlackey84, 2019-04-22:
@jchuckw - thanks for your help.  I did the check that you mentioned, the results are below:

pi@raspberrypi:~ $ cd raspi-sump
pi@raspberrypi:~/raspi-sump $ ls -la
total 44
drwxr-xr-x  9 pi root 4096 Apr 21 13:37 .
drwxr-xr-x 21 pi pi   4096 Apr 21 13:42 ..
drwxr-xr-x  2 pi root 4096 Apr 21 13:36 charts
drwxr-xr-x  2 pi root 4096 Apr 21 13:36 cron
drwxr-xr-x  2 pi root 4096 Apr 21 13:36 csv
drwxr-xr-x  2 pi root 4096 Apr 21 13:36 docs
drwxr-xr-x  2 pi root 4096 Apr 21 13:40 logs
-rw-r--r--  1 pi pi   5469 Apr 21 13:39 raspisump.conf
drwxr-xr-x  2 pi root 4096 Apr 21 13:37 sample_config
drwxr-xr-x  4 pi root 4096 Apr 21 13:36 web

jlackey84, 2019-04-22:
@jchuckw - I took a little initiative and changed the group, and permissions of the raspi-sump directory.  After I did this I ran the rsump.py command again - still no CSV file.  Here is the directory status now:
pi@raspberrypi:~/raspi-sump $ ls -la
total 44
drwxrwxrwx  9 pi pi 4096 Apr 21 13:37 .
drwxr-xr-x 21 pi pi 4096 Apr 21 13:42 ..
drwxrwxrwx  3 pi pi 4096 Apr 22 06:59 charts
drwxrwxrwx  2 pi pi 4096 Apr 21 13:36 cron
drwxrwxrwx  2 pi pi 4096 Apr 21 13:36 csv
drwxrwxrwx  2 pi pi 4096 Apr 21 13:36 docs
drwxrwxrwx  2 pi pi 4096 Apr 21 13:40 logs
-rwxrwxrwx  1 pi pi 5469 Apr 21 13:39 raspisump.conf
drwxrwxrwx  2 pi pi 4096 Apr 21 13:37 sample_config
drwxrwxrwx  4 pi pi 4096 Apr 21 13:36 web

thanks

alaudet, 2019-04-22:
Very strange.  Can we see the permissions in the csv folder only

```
cd csv
ls -al
```

jlackey84, 2019-04-22:
Hi, results below:
~/raspi-sump/csv $ ls -al
total 12
drwxrwxrwx 2 pi pi 4096 Apr 21 13:36 .
drwxrwxrwx 9 pi pi 4096 Apr 21 13:37 ..
-rwxrwxrwx 1 pi pi   80 Jan 27  2018 README.md

jchuckw, 2019-04-22:
Does not look like the permissions would be keeping it from working at the least.   I am not sure if group is supposed to be root after the install - but would not think so?  Regardless looks like user pi had r/w/x permissions even before you changed them all over to pi:pi.

I dont have any other great ideas unfortunately.  When I initially installed I had the same issue as your initial post.   Then I tried the longer install command  'sudo pip3 install --no-binary :all: raspisump'.   I still could not get it to work, so I just removed it with:
 sudo pip3 uninstall raspisump
Then - re-installed with:
sudo pip3 install --no-binary :all: raspisump

At this point it worked fine for me.

alaudet, 2019-04-22:
Ok, so here is what we know.

- running rsump.py seems to run and gives no error
- permissions on your folders look fine
- the issue appears to be nothing related to rsumpchart.py

I have made some assumptions

- you are running the python3 version  (you ran "sudo pip3" during install instead of simply "sudo pip")
- you are running the latest version of raspbian (stretch)

to keep it simple I would do the following.

- uninstall raspisump
`sudo pip3 uninstall raspisump`

- cp your conf file to a safe location

`cp /home/pi/raspi-sump/raspisump.conf /home/pi`

delete the raspi-sump folder

```
cd /home/pi
sudo rm raspi-sump -r -f
```

reinstall raspi-sump

`sudo pip3 install --no-binary :all: raspisump
`
copy conf file back to raspi-sump folder

`mv /home/pi/raspisump.conf /home/pi/raspi-sump`

Try running rsump.py again

`rsump.py`

alaudet, 2019-04-22:
finally if all that fails, try and take a reading with hcsr04.py

The command is as follows

`hcsr04.py -t <trigpinvalue> -e <echopinvalue>`

Copy paste the output..  This will test if your sensor is working.

jlackey84, 2019-04-22:
I did the uninstall/install procedure using the lines you posted, but still no CSV file.
Here is the output from the sensor test:
$ hcsr04.py -t <trigpinvalue> -e <echopinvalue>
bash: syntax error near unexpected token `newline'

alaudet, 2019-04-22:
you have to put your pin numbers.   I use pin 17 for trig and pin 27 for echo.  So I get the following;

```
hcsr04.py -t 17 -e 27
trig pin = gpio 17
echo pin = gpio 27
speed = 0.1
samples = 11

The imperial distance is 4.4 inches.
The metric distance is 11.1 centimetres.
```

jlackey84, 2019-04-22:
oops.  Missed that part.  Here's what I got:
$ hcsr04.py -t 17 -e 27
trig pin = gpio 17
echo pin = gpio 27
speed = 0.1
samples = 11

Traceback (most recent call last):
  File "/usr/local/bin/hcsr04.py", line 79, in <module>
    main()
  File "/usr/local/bin/hcsr04.py", line 70, in main
    raw_distance = value.raw_distance(sample_size=samples, sample_wait=speed)
  File "/usr/local/lib/python3.5/dist-packages/hcsr04sensor/sensor.py", line 90, in raw_distance
    raise SystemError('Echo pulse was not received')
SystemError: Echo pulse was not received

alaudet, 2019-04-22:
Now we are getting somewhere.  That error means that your raspberry pi is not communicating properly with your sensor.

Things to check
- make sure that you are using the actual gpio pins you are declaring.  
- if your pins assigments are confirmed correct then this is pointing to a faulty cable or sensor.
- you can try another cable or sensor to eliminate that possibility.

How did you make your wire?  What gauge did you use?  What resistors did you solder to your wire for the voltage divider?

The voltage divider is needed to bring down the voltage from 5v on the power pin of the pi to 3.3v which the sensor is rated for.  There is a diagram of my wire setup on https://www.linuxnorth.org/raspi-sump/

So forget about raspisump until you can communicate with your sensor with hcsr04.py.  That is most likely the root of the problem.

alaudet, 2019-04-22:
This link to the hcsr04sensor library may be of use to you as well.

https://www.linuxnorth.org/hcsr04sensor/

jlackey84, 2019-04-22:
That makes sense.  I followed the instructions for the voltage bridge on the Linuxnorth site - 470ohm R across echo wire, 1kohm R across pi-side of 470ohm R and gnd wire.  For cables, I used the connectors that came with the sensors, with 12ga wire between.  I checked pin-to-pin resistance (pi pin to sensor pin) and all look good, which tells me the socket connections are good:
All wires have good continuity across (<0.2ohm for 5V, trig, gnd; 465ohm for echo)
No shorts between 5V / trig and the others
Echo = 465ohm
Sensor-side echo to pi-side gnd = 1462ohm
Sensor-side gnd to pi-side echo = 996ohm
Also, I've tried 2 sensors with same results...

alaudet, 2019-04-22:
I use 24ga wire (CAT5) and my cable is about 4ft long.  The fact that you tried two sensor points to the cable or the pin assignments as the culprits.

Which model of raspberry pi are you using and which pins are you connected to?

jlackey84, 2019-04-22:
Pi model 3B V1.2
5V to pin 2
Gnd to pin 6
Trig to pin 17
Echo to pin 27
[image]

alaudet, 2019-04-22:
Then I do not know.  The software is not receiving a signal.  Its sending it out, but it's timing out waiting for the echo pulse to be received on echo.

This could mean its getting sent out the trig pin, but something is going wrong along the trig wire channel or echo wire channel to the moment it is received on the echo pin.

Try a different trig pin.  Then a different echo pin.   If still problems then I cannot see how it can be anything else except your cable.

alaudet, 2019-04-22:
By any chance, did raspi-sump create this error_log in /home/pi/raspi-sump/logs?

```
pi@bigpapi:~/raspi-sump/logs $ ls
alert_log  error_log  heartbeat_log  README.md
pi@bigpapi:~/raspi-sump/logs $ cat error_log 
2019-04-05 07:48:19,**ERROR - Signal not received. Possible cable or sensor problem.

```

jlackey84, 2019-04-22:
It did leave that log - 2019-04-22 14:50:02,**ERROR - Signal not received. Possible cable or sensor problem.
I'll try switching pins, and if that doesn't work I'll make up a new harness.  Thanks for the help!!
Jeremy

alaudet, 2019-04-22:
Wish I had asked you that first.  Would have save quite a bit of back and forth before zoning in on the signal issue.  lol.  Ah well...

jlackey84, 2019-04-24:
Solved!  I'm embarrassed to admit it, but maybe it will help someone else...  I was using the 1-40 pin numbers for the GPIO assignments, NOT their GPIO pin numbers.  It's working fine now.
Thanks again for all of your help and patience!!
Jeremy

closed 2019-04-24 by jlackey84

alaudet, 2019-04-24:
All good, glad you got it going.  :-)