alaudet/raspi-sump #37: ValueError: time data does not match format

github-t, 2019-04-27:
Out of nowhere I started getting the following error:

Traceback (most recent call last):
  File "/usr/local/bin/rsumpwebchart.py", line 25, in <module>
    main()
  File "/usr/local/bin/rsumpwebchart.py", line 21, in main
    webchart.create_chart(homedir)
  File "/usr/local/lib/python3.5/dist-packages/raspisump/webchart.py", line 36, in create_chart
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.5/dist-packages/raspisump/todaychart.py", line 54, in graph
    converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 985, in loadtxt
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 985, in <listcomp>
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "/usr/local/lib/python3.5/dist-packages/raspisump/todaychart.py", line 45, in bytesconverter
    return strconverter(s)
  File "/usr/lib/python3/dist-packages/matplotlib/dates.py", line 287, in __call__
    return date2num(datetime.datetime(*time.strptime(s, self.fmt)[:6]))
  File "/usr/lib/python3.5/_strptime.py", line 504, in _strptime_time
    tt = _strptime(data_string, format)[0]
  File "/usr/lib/python3.5/_strptime.py", line 343, in _strptime
    (data_string, format))
ValueError: time data '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0013:58:01' does not match format '%H:%M:%S'

I haven't had the chance to dig in any deeper.  Just curious if anyone may have run into this.

alaudet, 2019-04-27:
Yes, I have seen that.  If you open up your csv file find the line that has \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0013:58:01'  .    Delete that line and save the file.  This happens on corrupted readings very rarely.   It's happened to me 3 times over the past four years.  After the line is deleted the graph will be able to be created.

closed 2019-04-30 by alaudet

github-t, 2019-04-30:
That was it.  Thanks!