alaudet/raspi-sump #38: Rolling Charts

dolszews89, 2019-05-01:
I am looking to generate a rolling chart that I can add to my Raspi-Sump webpage, something in the range of a 4 hour window.  Has anyone here attempted to do this previously?  I've look into modifying the current "rsumpwebchart.py" code and the "webchart" backend is currently a bit above my skill level.  I just wanted to check if anyone has done this before I jump in!

spankurmonkey, 2019-05-01:
Check out this fork -- I have modified the page.. it's not rolling but I think its quick and easy to use:

https://github.com/spankurmonkey/raspi-sump

Also, I have a modified script that feeds into ThingSpeak -- see my public feed here:

https://thingspeak.com/channels/737963

I have a widget install on my cellphone that reads right from Thingspeak so I have real-time data on my mobile if you will.....

alaudet, 2019-05-02:
so you always want to show a chart of the last four hours only?

dolszews89, 2019-05-02:
@spankurmonkey Thanks for the reply.  I like the widget idea.  I'll take a look at your fork.

@alaudet Yes, but 4 hours was just an idea.  Ideally, I guess, it could be any amount of time user configurable (2,4,6 hours, etc.).  We had a lot of HEAVY rain last night and my pump has been cycling like crazy which made the normal graph hard to read (see graph).  (Fun note: the flat line was my pump running non-stop because of the amount of water flowing into my basin matched what my pump could output.  I though my detector was faulting).

[image]

That's what I was think of doing, if you could offer any advice I would greatly appreciate it.  I'm not too familiar with Python so this will be an educational adventure!

alaudet, 2019-05-02:
Your pump is working as hard as mine these days.
Charts could use some attention for sure.  I have had requests to be able to chart multiple CSV files, make them look different, all kinds of things.  I have some ideas for charting but all that is lacking is time to work on it.

For a rolling chart I would see simply giving the option when running the chart script to tell it how many readings it should graph.  It's do-able but I really don't know when I will be able to get to it.  I will enter it as a feature request in the tracker, but that doesn't mean I can get to it quickly.

alaudet, 2019-05-02:
Closing this issue and rolling it into issue # 39 for a overhaul of chart reporting capabilities.

https://github.com/alaudet/raspi-sump/issues/39

closed 2019-05-02 by alaudet

rescuemoe, 2019-05-17:
spankurmonkey, would you happen to have any info on how you set that up on your phone
That is really cool 
Thanks