alaudet/raspi-sump #39: Overhaul chart reporting

alaudet, 2019-05-02:
I am opening this issue as a feature request.  There have been requests to provide more options with chart generation;

- allowing user selected colours
- graphing charts based on old data
- graphing specific days
- rolling charts that graph defined intervals.

This issue will track progress on chart options and other issues will be closed.

sregiT, 2019-05-15:
Al,

I’m going to try to find some time to look into D3.js https://d3js.org/. I think I can figure out how to read in the cvs files with php (maybe using jquery also) and create dynamic charts. This should be less cpu intensive for the pi also because the charts will be rendered in the browser instead.

alaudet, 2019-05-15:
I think it would be an interesting as an add-on.  I really know nothing about php/jquery so would have trouble maintaining it.  Not sure I could roll it into master.  Like to see what you come up with though.

MarkPinches, 2019-06-02:
Guys it would be pretty straight forward to create small shell scripts to create data sets of different size length and granularity to support a simple extensions to the front end. e.g. an daily all time, a 6mtd, 3 mtd, ytd etc.. These could have different output .csv files that could be pointed at in the charting function. see my other issue as an example.

alaudet, 2019-06-02:
MarkPinches I am going to create a  branch called charting to start throwing around some ideas there.  Any work toward modifying charting will be done on that branch specifically.

alaudet, 2019-06-09:
**devel branch update**
added [charts] section to conf file.
Can now change line color of chart.

USAFPride, 2019-07-14:
I'm new to this project, but have you thought of using a database instead to store the data vs. CSV? A simple RRD (or you could go full out with sqlite3) would allow easy storing, mapping, etc.

alaudet, 2019-07-14:
Yes I have considered sqlite3 and will probably go that way.  I have to brush up on that before doing so.  I have a few changes planned that will coincide with the dropping of Python2 and Wheezy support.

USAFPride, 2019-07-14:
I can help with the DB portion.  I'll start roughing some items up in sqlite3. I have another project that I can pull dbase creation and writing from.

alaudet, 2019-07-14:
i appreciate the offer for help.  Before getting too far though I would like to consider adding a db for version 2 of raspisump.   I had already planned to start adding other features.  I was looking to have it coincide with dropping support for Python2 in January.

I think what I will do is start a new thread for version2.  I will also add a label for RaspiSump2 for all issues that need to be tracked.

By all mean try some proof of concept code but db support will be targeted for version 2.

I am going to add the thread with all features I am looking to add and provide guidelines for contributors so that anyone who wants to contribute has a framework to work within.

alaudet, 2022-02-12:
Since this thread has died I have added extra chart reporting capability to version 2.  If there is interest I can revive.

closed 2022-02-12 by alaudet