alaudet/raspi-sump #40: Relay support

MathieuMorrissette, 2019-05-03:
Add support to trigger a relay using GPIO pins if the water level is critical.

This could be used to trigger a 12v backup pump.

A long running python script could be triggered that turn on the relay and polls multiple times per minutes the water level and when it reaches a defined safe level it would turn off the relay and therefore the backup pump.

alaudet, 2019-05-03:
Interesting idea but wouldn't a float switch be more reliable than a pi to trigger the pump?  Or am I missing something.  The float switch would trigger at a slightly higher level than the main pump.  I could see the program sending an alert that the backup pump is most likely running.  But I am not sure I understand the value in having the pi trigger the pump when float switches are more likely more reliable than a raspberry pi.

gordee78, 2019-05-03:
I’m completely in agreement with Al here, and in fact this exactly how my setup is today.  

One wants the 12V backup pump system to run independently of the arguably less reliable Pi system (a float switch is easily tested, replaced, independently).  As long as the float is placed higher than the main pump float switch, you have all the benefits of Pi reporting/alerting/charting with the backup pump in place - without having Pi being the critical functional path. 

Gordon

alaudet, 2019-05-03:
Yes, I see raspisump as a passive monitoring system independent of the actual pumping system.  I really would not want to make part of the active pumping system.  That being said @MathieuMorrissette , I would be interested in seeing any code and/or pics of something you have done with a Raspberry Pi that implements this sort of functionality.  I could see uses for it in less critical systems.

I appreciate the new ideas, makes me think of possibilities.  Cheers.

MathieuMorrissette, 2019-05-03:
Alright I will try to come up with something on my side. I understand that this should stay a passive monitoring system.

The backup pump was just an idea and didn't think about the implications. 

A better idea would we to turn a buzzer or some kind of indicator that is unrelated to the pumping mechanism.

I might work on adding some kind of REST api so it's easier to integrate with other systems.

I'll close this issue.

Best regards

closed 2019-05-03 by MathieuMorrissette