alaudet/raspi-sump #41: Mesasurement Spike and stays High

rescuemoe, 2019-05-10:
So I finally got a around to wiring everything to the Pit, and since I have a lot in the pit, Main pump and back up pump I was get bad readings because I have limited open space.
So I took a 3" PVC pipe and mount the HC-SR04 sensor in the pipe pointing strait down.
It sits about 25" from the bottom of the pit. Things where working great for about 1 week. The range with the water is about 5" to 8". I have it set to read ever 2 mins.  All of sudden I started to get reading of 22". The CSV file would have 7.5 7.6 then jump to 22" and stay like that. Some times it would go back down for a few reads.
Has anybody ever encountered that before? The version I am running is from an install back at the end of 2018.  I swapped out the sensor as I purchased a 5 pack and it did the same think but this time it start about 4 mins after putting on the new sensor
Second question since I am using a version from 2018 would love to use some of the new features you added. How do I go about to get the latest code for the program on the PI?  DO I just run sudo pip3 install -U raspisump
THanks

jchuckw, 2019-05-10:
I have had similar issues.  I used a 4 inch PVC myself, and it is very likely to be one of 2 things.  

1 - there is some kind of debris that has floated up with the water, and stuck to the inside of the PVC.  If it is this, you can clean the pipe out fully, then put something simple like a pair of old pantyhose or something over the bottom end of the PVC. 

2 - It can ALSO just be water droplets.  This only happened to me once, when I had a power outage that caused the water to get higher in the pipe then normal.   For this, just clean the inside of the pipe.  I have not had to, but I would think if it is just water droplets, you could use something like rain-X on the inside of the pipe to keep the water from beading up.

Essentially, when doing it in a pipe, the sensor is much more sensitive than you think it should be :-)   That is at least what I found out.

spankurmonkey, 2019-05-10:
i am using a 2" pvc pipe and have not had any spikes.  Same sensor as last year.

rescuemoe, 2019-05-10:
Thanks for the advice. I cleaned up pipe and removes some debris at the bottom so far so good. going to have to come up with a way to keep the pipe more closed off but let the water run through it

closed 2019-05-10 by rescuemoe