alaudet/raspi-sump #43: Report request Pump Cyles

rescuemoe, 2019-05-17:
Just a thought for stats: when the Graph gets calculated also calculate how many times the Pump Cycled and keep a running total for the day

spankurmonkey, 2019-05-17:
@rescuemoe 
Not sure if you seen this option using ThingSpeak -

Raspi-Sump feed into ThingSpeak - tracking sump water level, number of sump cycles, minutes since last sump cycle and gallons discharged 

https://thingspeak.com/channels/737963

[image]

spankurmonkey, 2019-05-17:
@rescuemoe 

Also on my mobile too:
[image]

rescuemoe, 2019-05-17:
Ya I would love to do this.
Are there any instructions on how to set this up?

alaudet, 2019-07-05:
The reason I have not added it is because every pump is different. I have been using a local application that does this but its not consistent.  All I have are readings so I need to identify the pump cycling by when it records a series of readings that are decreasing in depth.

It all falls apart in the spring when my pump kicks off  a minute after emptying as the readings get messed up.

Would need a way to hook into the pump to see when it triggers but its really not something I have considered.

closed 2019-07-05 by alaudet