alaudet/raspi-sump #44: Changing plot to show multiple days

MarkPinches, 2019-06-02:
Hi,

Firstly Thanks so much for raspisump, it saved me a heap of time setting up a monitor for my rainwater tank. But with rainwater tanks you are interested in seeing a graph of the tank over longer periods than a single day as is the default set up for raspisump. So I have written a little shell script that generates a new file of readings for each day using the last line of the daily csv. Its then a fairly simple ( a couple of lines ) change in the raspisump code to display this.

Firstly the shell script... saved into a new folder (scripts) in raspi-sump dir. Call it rsump.sh
```
#!/bin/bash

DIR=/home/pi/raspi-sump/csv

rm "$DIR"/daily.csv

for FILE in "$DIR"/*; do 
    if [[ "$FILE" =~ ^.*waterlevel.*  ]]; then  	
		SUBSTRING=$(echo $FILE| cut -d'-' -f 3 | cut -d'.' -f 1)
		DATE=$(date -d $SUBSTRING +'%Y-%m-%d')	
        TAG=$( tail -n 1 "$FILE" | cut -d',' -f 2)
        STR=$DATE","$TAG
        echo $STR >> "$DIR"/daily.csv
    fi 
done
``` 
This creates a daily.csv in the csv folder.
You then have to add the shell script to the picrontab file. You need the full path to the file in the scripts dir, like...

```
59 * * * * * /usr/local/bin/rsumpwebchart.py
00 * * * * * /home/pi/raspi-sump/scripts/rsump.sh
```
Now for the changes in the raspisump package. These files are located in...
`/usr/local/lib/python3.5/dist-packages/raspisump
`
First webchart.py change the create_chart function to point to the daily.csv file and amend the bytedate2str format to reflect dates rather than times.

```
def create_chart(homedir):
    '''Create a chart of sump pit activity and save to web folder'''
    csv_file = '{}csv/daily.csv'.format(
        homedir
        )
    filename = '{}charts/today.png'.format(homedir)
    bytes2str = todaychart.bytesdate2str('%Y-%m-%d')
    todaychart.graph(csv_file, filename, bytes2str)
```
And in todaychart.py add in ...
`date = mpl.dates.num2date(date)`
After the numpy.loadtxt
And thats it. Only tested on python3. Cheers!

alaudet, 2019-06-02:
Very cool.  This is something I would consider merging in.  The script would reside in the bin directory and be installed to /usr/local/bin

For the changes in the packages it's a matter of allowing a variable to be sent instead of hardcoding the filename.  So something like;

`csv_file = '{}csv/{}.csv'.format(homedir, csvfile)
`
The script should pass the csvfile variable.

It's been a while since I looked at matplotlib...what is the reason for 

`date = mpl.dates.num2date(date)`

I just have to go read the documentation.  I will try what you provided though.

alaudet, 2019-06-02:
Any chance of creating a pull request off the devel branch?  We could work out the details on a new branch with the goal of merging to master after making sure everything works as intended for users not needing the functionality. 

I know of someone using this for monitoring a drinking water cistern and different graphing might help.

alaudet, 2022-02-12:
Nothing really came of this.  I will close this discussion but will most likely come back to this when I start working on charting.

closed 2022-02-12 by alaudet