alaudet/raspi-sump #45: Systemd config for raspi-sump

edsai, 2019-06-02:
I wanted to run the script raspi-sump constantly and have systemd start it up, restart it, and report errors if the script wasn't running instead of hoping cron would take care of things. I still had the chart creation done in cron. This may save some other people time and I'm happy to do a pull request with the contents of the systemd service file below.

Short instructions for systemd-based systems:

1. Make sure reading_interval in raspisump.conf is set to something other than 0

2. Create /lib/systemd/system/rsump.service with the following contents:

[Unit]
Description=raspi-sump service
After=multi-user.target

[Service]
Type=idle
SyslogIdentifier=raspi-sump
ExecStart=/usr/bin/python3 /usr/local/bin/rsump.py
Restart=on-failure
RestartSec=60s

[Install]
WantedBy=multi-user.target

2. sudo systemd enable rsump
3. sudo systemd start rsump

alaudet, 2019-06-02:
Hi, makes sense.  I have never liked the current solution and would like to test this approach.

Please issue all pull requests off the devel branch.  Cheers

edsai, 2019-06-02:
Where would you like the systemd file to go in your repo? cron doesn't seem like a place it should go.

alaudet, 2019-06-02:
Create a folder called systemd.  If you could add some instructions/information on what is going on that would be helpful.  My systemd knowledge is lacking.  What I need to do to test is see which parts of the current system get ripped out.   How hard it is to implement for users who have no Linux knowledge.  How hard it is to stop the service.  

Is it simply change interval back to 0 and stop the service? Reboot?

Mostly exactly what behaviour is happening when someone uses this approach.

alaudet, 2019-06-02:
This also has to work for people still using Raspbian Wheezy which is getting old.

edsai, 2019-06-02:
Yes, this will work with Jessie and newer but unfortunately not Wheezy. Everything after Wheezy transitioned to systemd. Nothing gets ripped out other than changing the interval to something non-zero and commenting out the cron job for rsump.py itself or using the systemd method if you haven't created the cron job in the first place. It eliminates the possibility of a race condition if you set the interval low and use the cron hack of setting up multiple jobs with sleep functions and hoping there won't be a case where one cron job hasn't completed for some reason and the next one fires off.

edsai, 2019-06-02:
BTW, awesome work. I'm still impressed you've stayed involved with the project. I set up everything a few weeks ago but wanted to dump data into InfluxDB + Grafana. At the moment I'm tinkering with node-red and have it dump directly to InfluxDB. What I really wanted was mqtt functionality which I suppose I could add to the project but node-red gives me some flexibility to display sump water level on an LCD and monitor a backup sump battery voltage with an ina219 sensor.

Your all-in-one project is really great for simplicity. I also had my raspi-sump setup sending to a pushover email alias so I could get push notifications on my phone. Like I said, awesome work.

alaudet, 2019-06-02:
Thanks, as long as I own this house I have a vested interest in waterlevel monitoring.  :-)

alaudet, 2019-06-02:
Actually what I will probably end up doing is issuing a logged deprecation warning for people using the old method.  Time to start getting away from wheezy (which includes me).

alaudet, 2019-06-02:
I am looking to add a battery powered backup pump.  My current backup pump is powered with a generator, and not the automatic start on power failure kind.  How long can your pump function on battery power alone?  Interested on the details of you monitoring.

edsai, 2019-06-02:
My battery backup pump is a bit of an odd bird. It's a 24v backup system with 2x12v 75Ah batteries wired in series made by Stevens Pumps which is actually out of business now. Going by the math and the wired pump cycle time (45sec at 50gpm) and it usually cycles 1 once per hour under heavy rain, the batteries would hypothetically last for 430 hours if it had to cycle 2 times an hour. I was oversold but at the time I just wanted a backup pump system and I overpaid for a way-oversized system. The batteries alone are $175-$250 a piece. I looked into getting a new system with monitoring but it's hard to dump money into an inferior pump with a smaller battery system and a proprietary alerting system which led me down the path of raspi-sump.

I'm still putting things together but so far I've got node-red using the pi-srf module to read from the JSN-SRT04 sensor that I was using with raspi-sump. I'm using a buck converter to power the pi off the 24v batteries (since they're constantly charged) going directly into the microusb port so I don't have to deal with a dedicated battery backup for the pi. I've got an hd44780 lcd showing the water level. I'm waiting on a bench power supply so I can do some testing with an ina219 voltage sensor so I can monitor battery power. I'm not an electrical engineer and I want to make sure I don't do something with the ina219 that can blow the board. 

These are all things that you could do in raspi-sump but node-red is super easy and very expandable so I'm using that for now. My vision really was to have all of this integrated into raspi-sump, build a docker image of raspi-sump and do an automated deployment of said image by using hypriot on the pi and doing an automated deployment with ansible. There are other easier ways to deployment with things like balena.io's platform to just deploy images to the pi and you're ready to go. Flash the balena.io image, it talks to the cloud, the cloud installs the image you tell it to on the pi with the container/app you want and you're ready to go. The downside is that those platforms are meant for large scale IoT and limit the user on the number of images they can do for free. Containers remove all of the dependency issues and manual installation of packages etc on the pi. I could still probably put a docker image out there for raspi-sump and people could deploy it themselves with one of the many IoT platforms out there. The pi plus what you've built has a ton of possibility and the sump pump and monitoring companies out there are so much further behind in terms of their services because their focus is on building pumps, not software.

gordee78, 2019-06-02:
FWIW...

This backup system seems over powered for the need. You can probably get cheaper 12V lower current batteries to shave cost. 

My sump pump easily runs every 20-3 mins on dry summer months, or every 5-10 mins during spring thaw (Toronto, Ontario). 

I use a 12V liberty pump system as a backup to a AC pump.  https://www.libertypumps.com/Product/Model-441

And like you I power my raspisump directly from the 12V battery using an non isolated 9-48V power supply.   The battery can last longer if needed I I hook up additional batteries or ... if necessary ... a car in parallel. Because of Al’s contributions here and the backup sump I think I have an extremely reliable system. 

G

alaudet, 2019-06-02:
@edsai  I like docker but hadn't seen a need to use it for raspisump because its so easy to deploy and is pretty forgiving on python versions.  So any board with GPIO pins running linux with python should be able to interface with the sensor.

You guys are showing me I need to update my battery backup game.  That's my next project and really like what @gordee78 has done which we  have discussed before.  I appreciate all the great info and ideas.

edsai, 2019-06-02:
@alaudet docker is just appealing from the stateless nature for me. If something happens to the pi (corrupt SD card, whatever), I want to make it as easy as possible to re-provision.

@gordee78 good point, I can always get smaller 12v batteries. They're about 5yrs old so I'm thinking they'll probably need replacing soon. 

I forgot to mention that I've got a smart plug I flashed with sonoff-tasmota so I could monitor current draw and cycle frequency of the main pump. It's dumping data via mqtt to mosquitto and that's being picked up by telegraf/InfluxDB and being fed into Grafana.

alaudet, 2022-02-12:
closing this discussion as I will move to systemd for restarting the service on Version 2 of Raspi-Sump.  Added to V2 feature list.

closed 2022-02-12 by alaudet