alaudet/raspi-sump #46: Liters Value

bjornarfonn, 2019-06-03:
Is there anyone who can help out on making a line of text on the chart that states Liters of volume.

If there is a way to add radius in the PIT config, we can calculate the following;
(3.14*r^*H) - total volume
(3.14*r^*H) - measured volume
(total volume)-(measured volume) = current pit volume

alaudet, 2019-06-03:
So what you are asking is for raspisump to give the option to measure volume instead of waterlevel and to graph the watervolume over time.  It's possible but I am trying to understand use cases.  At its nature raspisump is concerned with high water levels to prevent flooding.  Water volume sounds like a different use case.  What is your use case for watervolume?

As for if it is possible,  anything is, the question is if it belongs there.

bjornarfonn, 2019-06-03:
I have been trying to find a suitable application to measure a water well at my cabin. Its currently being done by a measuring stick. I could use your application "as is" and just do the math myself. But would be nice to get a current volume readout on the report.

I would not be looking at changing the chart as they are now, but perhaps a small additional field to the chart that would add a "current pit/well/sump water level (Liter)" in addition to the graph. My question is more directed towards;  With some few additional inputs in the config and then a formula to combine it all to the correct output, can this be done without changing the current functionality. 

RADIUS(r) | 30 | Cm | [PIT] 
TOTAL HEIGHT (h) | 400 | Cm | [PIT] 
MEASURED HEIGHT (mh) | 210 | Cm | [PIT] 
SENSOR DISTANCE (d) | 10 | Cm | [SENSOR] 

VOLUME (v) | 565.2 | L | [CHART] 

FORMULA | ((3.14*r*r*h)-(3.14*r*r*(mh-d)))/1000

alaudet, 2019-06-04:
Maybe in a future version I could incorporate it, but it goes beyond what the program is meant for which is basically for alerting.  However the source code is open and anybody can fork the repository and add the functionality...and even distribute their changes. I am not saying no, but would be willing to look at code contributions from others wanting to add it.  I could possibly merge it in at that time after testing it.  But it is not on my radar to lead it.

alaudet, 2019-06-30:
Just an update, I am adding functionality to the hcsr04sensor library to calculate liquid volume in various types of containers (cylinders, boxes, elliptical tanks).  This is the first step in providing what you are looking for.  I am not sure I would add it to raspisump but may look at another application that can do something similar.

No promises.  I am sure you could find someone that could script something together for you fairly easily using that library.

bfonn, 2019-07-01:
Thank you for that! I'm sure I can get someone to help me produce a Volume display on the index site. Will also let you know that raspi-sump has now been installed at our cabin, monitoring the water well. Greatly appreciate the excellent software!

alaudet, 2019-07-01:
That's awesome, thanks for letting me know that.

I will let you know when I complete the additions to hcsr04sensor.  I still need to clean things up, write tests for it and update the documentation on how to use it.

alaudet, 2019-07-02:
I will be updating progress at the issue tracker for the library.

https://github.com/alaudet/hcsr04sensor/issues/14

alaudet, 2019-07-07:
What is the shape of your well.  Is it a perfectly round cylinder shape?

alaudet, 2019-07-07:
Making changes to hcsr04sensor module.

This script does what you need

    from hcsr04sensor import sensor
    from hcsr04sensor import sensor

    trig_pin = 17
    echo_pin = 27

    # Get litres in a cylinder
    cyl_depth_metric = 410  # centimeters from sensor
    cyl_radius_metric = 30  # centimeters
    value = sensor.Measurement(trig_pin, echo_pin)
    distance = value.raw_distance()
    #water_depth_metric = value.depth(distance, cyl_depth_metric)
    water_depth_metric = 410 - 210
    volume_litres = value.cylinder_volume_standing(water_depth_metric, cyl_radius_me$
    print("The liquid volume of the cylinder is {} litres".format(round(volume_litre$

When I ran this with your values I got the following.

pi@bigpapi:~ $ python3 ./bfonn.py 
The liquid volume of the cylinder is 565.5 litres

Here is a direct link to the script that you can test with your sensor.

https://github.com/alaudet/hcsr04sensor/blob/devel/recipes/cylinder_volume_metric.py

As you can see I commented out the water_depth_metric and hardcoded your value of 410 - 210 (depth of the water in the well).

You can use actual water_depth_metric line to get live readings of your well.

I know this doesn't provide graphing but it is the first step in what you are looking for.

bfonn, 2019-07-07:
The well is cylindrical. Consistant all the way to the bottom

alaudet, 2019-07-07:
I figured that after I plugged in the numbers into the function to calculate volume of a cylinder.  It came exactly to the number you calculated. 

To calculate the volume you need

volume = pi * radius * radius * water_depth

The following method in hcsr04sensor is used.

    def cylinder_volume_standing(self, depth, radius):
            """Calculate the liquid volume of a standing cylinder"""

            volume = self.pi * radius * radius * depth
            if self.unit == "metric":
                 return volume / 1000
            else:
                 return volume / 231

closed 2019-07-11 by alaudet