alaudet/raspi-sump #47: Raspbian Stretch Lite

bjornarfonn, 2019-06-09:
Have tested raspi-sump for a while now, and has been working great. Tried installing it a smaller and more lightweight build today using Raspbian Stretch Lite, but that doesn't work. 
Time formatting doesn't work, graph line is not displayed and the graph doesn't update with cronjob. Is there a known issue with some dependencies and Raspbian Lite?

Raspberry Pi 3 
Python version 3.5.3
Running lighttpd on the same PI
[image]

alaudet, 2019-06-09:
I have always used Raspbian Lite .  Did you use no-binary all when
installing?  See the install doc.

bjornarfonn, 2019-06-09:
Yes, installed as per install doc. Formatted the sd card and re-installed everything again, still the same issue? Anything you could point me too in terms of troubleshooting?

bjornarfonn, 2019-06-09:
So after some trials and errors, it seems that it failed to start running the process? I tried changing /etc/rc.local adding sudo;
sudo /usr/local/bin/rsump.py &

It seems to be producing charts now, will try some more testing. Is this something you have any experience with?

alaudet, 2019-06-09:
usually with stretch you don't need to run sudo.  you def don't need to run it from rc.local as you are already starting the process with elevated priviledges at that point.

What happens if you kill the process and then try and start it without.

sudo killall 09 rsump.py
/usr/local/bin/rsump.py &

bjornarfonn, 2019-06-09:
hmm... i spoke to soon when i said i got it running. It did one graph and that was it. 
removed the sudo as your comments states, its starting with privileges. 

Trying to killall gives me this output;
pi@raspberrypi:~/raspi-sump/logs $ sudo killall 09 rsump.py
09: no process found
rsump.py: no process found

Starting it manually using /usr/local/bin/rsump.py &
then trying to generate new graph by doing $ rsumpwebchart.py does nothing.

Starting it manually and stopping it again gives me this output;
pi@raspberrypi:~/raspi-sump/logs $ sudo killall 09 rsump.py
09: no process found
rsump.py: no process found
[1]+  Done                    /usr/local/bin/rsump.py

My logs folder is also empty, so there is no help there

alaudet, 2019-06-09:
try running it not as a process to start.  Simply run rsump.py

Does it create a csv file in /home/pi/raspi-sump/csv?

What subfolders are in you /home/pi/raspi-sump/csv file?

bjornarfonn, 2019-06-10:
It works when adding rsump.py to crontab with 1minute intervals. It's a weird issue, as basically the startup file is doing the same thing, the difference being its starting as a continual process rather than starting occasionally.

So my crontab is running rsump.py every minute, that produces a sample from the sensor every minute, and then I am generating a chart every 10 minutes (will change this when deployed outside) this all seems to work. So, I guess I'm happy with this so far.

*/1 * * * * /usr/local/bin/rsump.py
*/10 * * * * /usr/local/bin/rsumpwebchart.py
*/5 * * * * sudo /usr/local/bin/rsumpmonitor.py

alaudet, 2019-06-10:
If you launch it as a process manually does it work as intended?  Or is it
just when you use rc.local?

closed 2019-07-05 by alaudet