alaudet/raspi-sump #49: Error when running rsumpwebchart.py after upgrade to Raspbian Buster

github-t, 2019-07-09:
When I run:
`/usr/local/bin/rsumpwebchart.py`

I get:
`
/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py:424: MatplotlibDeprecationWarning: 
Passing one of 'on', 'true', 'off', 'false' as a boolean is deprecated; use an actual boolean (True/False) instead.
  warn_deprecated("2.2", "Passing one of 'on', 'true', 'off', 'false' as a "
Traceback (most recent call last):
  File "/usr/local/bin/rsumpwebchart.py", line 25, in <module>
    main()
  File "/usr/local/bin/rsumpwebchart.py", line 21, in main
    webchart.create_chart(homedir)
  File "/usr/local/lib/python3.7/dist-packages/raspisump/webchart.py", line 36, in create_chart
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.7/dist-packages/raspisump/todaychart.py", line 100, in graph
    plt.savefig(filename, dpi=72)
  File "/usr/lib/python3/dist-packages/matplotlib/pyplot.py", line 689, in savefig
    res = fig.savefig(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/figure.py", line 2094, in savefig
    self.canvas.print_figure(fname, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/backend_bases.py", line 2075, in print_figure
    **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/backends/backend_agg.py", line 521, in print_png
    cbook.open_file_cm(filename_or_obj, "wb") as fh:
  File "/usr/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 407, in open_file_cm
    fh, opened = to_filehandle(path_or_file, mode, True, encoding)
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 392, in to_filehandle
    fh = open(fname, flag, encoding=encoding)
PermissionError: [Errno 13] Permission denied: '/home/pi/raspi-sump/charts/today.png'
`

alaudet, 2019-07-11:
Must be something during install.

Do the following.

    sudo pip3 uninstall raspisump  
    sudo pip3 install --no-binary :all: raspisump

uninstalling will preserve you raspi-sump folder in /home/pi and your conf file.

alaudet, 2019-07-12:
assuming it worked so closing the issue.  reply to this issue if you want to reopen.

closed 2019-07-12 by alaudet

ctktfg, 2019-07-17:
Sorry for the delay.  After running the commands above, I now get the following error:

```
/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py:424: MatplotlibDeprecationWarning: 
Passing one of 'on', 'true', 'off', 'false' as a boolean is deprecated; use an actual boolean (True/False) instead.
  warn_deprecated("2.2", "Passing one of 'on', 'true', 'off', 'false' as a "

```

alaudet, 2019-07-17:
That is just a deprecation warning from Matplotlib, the graphing library used for creating the charts.  Its saying that code should start using True/False statements instead of on/off ones.   Buster uses a later version of matplotlib.  The code still works though, and your chart should be created.
I will make changes to the charting code in the future to use True/False statements instead of on/off but the deprecation warnings will not affect output for now.

github-t, 2019-07-25:
So the problem is, this is run from a cron job and the output of the cron job is emailed if there are any warnings or errors.  So when it throws this warning, sends an email each time.  I could turn off emails but then the rest of my cron jobs would suffer.  Is there any way to point me to the type of code I should be looking for and I'll just change it to a true/false statement (assume that I don't know Python but that I can hunt through code and do a search and replace with the best of them :) ).

alaudet, 2019-07-25:
can you copy your cron entry?

github-t, 2019-07-25:
Oh that's a great idea.  I did that and everything is good.  Also I won't get those rare emails when it gets a screwy reading from the sensor.  Thanks (again)!

alaudet, 2019-07-25:
Cheers :-)