alaudet/raspi-sump #50: AttributeError

jrreich, 2019-07-11:
Recently upgraded and now I get the following error when running rsump.py

Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 24, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.7/dist-packages/raspisump/reading.py", line 76, in water_depth
    water_depth = water_reading()
  File "/usr/local/lib/python3.7/dist-packages/raspisump/reading.py", line 67, in water_reading
    return round(value.depth(raw_distance, pit_depth), 1)
AttributeError: 'Measurement' object has no attribute 'depth'

alaudet, 2019-07-11:
It's almost like hcsr04sensor did not update along with raspisump.  There is a change in hcsr04sensor that now uses the depth method instead of either depth_metric or depth_imperial.

Can you try the following?

sudo pip install -U hcsr04sensor

if that does not work, do the following

sudo pip3 uninstall raspisump
sudo pip3 uninstall hcsr04sensor

sudo pip3 install --no-binary :all: raspisump

You will not lose your conf file or past csv readings when you uninstall raspisump.

alaudet, 2019-07-11:
What I provided you should work.  But I have made an update to setup.py to pin raspisump to dependency hcsr04sensor >= version1.6

jrreich, 2019-07-11:
Thank you much. I thought I tried updating hcsr04sensor without success, but it is working now after it upgraded to 1.6.1. It was previously at 1.5.3 somehow.

closed 2019-07-11 by jrreich

alaudet, 2019-07-11:
glad you got it going.