alaudet/raspi-sump #51: RaspiSump Version 2 Main feature list

alaudet, 2019-07-15:
This is the launch of RaspiSumpV2 development.

- Support for Python3 only.  Targeting v3.7+

If you have ideas for things you would like in version 2 please feel free to comment on this thread.

**Planned feature list**

- Web front end for managing RaspiSump
- Support for Sqlite3. (will be able import all your past csv files)
- Nicer graphs and more options for reports
- Support for cellular so that it can be used wherever there is cell coverage

**Contributing**

- A main branch will be created for version 2 of raspisump.  A devel branch will also be created.  Sub branches will be used off of the v2 devel branch only

- Nose will be used for testing.

- All issues related to RaspiSumpV2 will use the label that was created for it.   Other labels will be used to classify issues (Feature Request, Bug, Question etc etc)

spankurmonkey, 2019-07-15:
I understand charting overhaul and migration to DB vs the CSV file... but how about making an IOT feed into like ThingSpeak?

alaudet, 2019-07-15:
The main reason I don't bundle outside services right into raspisump is
because I want to keep the code independent of third party services that
require sign up.  It is for that exact purpose that I release the code
under the MIT license.  Anyone is free to take the code and extend it as
they wish, and even package and sell it if they want to and even keep their
code private.  As long as they adhere to the MIT license which is very
permissive and they don't call their app raspisump.

I would love if someone forked it and did cool things with their own
release.  For my purposes I will continue to offer the base package that
can be used, trying to improve it within the python language and it's
modules.

TheLazyIt, 2020-03-04:
As far as nicer looking graphs go, maybe Grafana could be of some help :) Use that with Zabbix or Nagios, and you could maybe get some additional reporting functionality. I don't know. I'm not a programmer, and have just learned that these tools even exist, but it seems to me like they would work for what you need.

alaudet, 2020-03-04:
Thanks for the feedback, it still comes back to my point above.   These are nice services for sure but I am hesitant to add dependencies to external services whose terms may change.

Graphing is something that can be easily implemented by someone else as well if they want to add this extra functionality.  For an open source application such as this I prefer to stick with well known python libraries.

The nature of the MIT license allows anyone to implement a preferred solution for graphing without any restrictions, it can be totally independent from Raspi-Sump.

alaudet, 2023-06-22:
This roadmap will be moved to the raspisump website

closed 2023-06-22 by alaudet