alaudet/raspi-sump #57: Confusion about where to find the conf file.

Arcade-Anima, 2019-09-22:
I am stuck at this portion of the install. 
cd /home/pi/raspi-sump
mv sample_config/raspisump.conf

I can find the raspi-sump files in usr/local/lib/python3.7/home/pi/raspi-sump  I even tried to move the raspisump.conf file through the gui but it failed to work.  Any thoughts?

alaudet, 2019-09-22:
The conf file is in /home/pi/raspi-sump/sample_config

Arcade-Anima, 2019-09-23:
ya but when I go to input the commands it says that there is no directory.

Arcade-Anima, 2019-09-24:
It might just be that I used Buster instead of Strech.

alaudet, 2019-09-24:
Should work the same on either one.

michael3975, 2019-10-05:
I'm having the same issue. When installing it's not copying the files to the /home/pi folder - It simply leaves the files in the /usr/local location.

pi@raspberrypi:~ $ pip3 install raspisump
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: raspisump in ./.local/lib/python3.7/site-packages (1.2.3)
Requirement already satisfied: hcsr04sensor>=1.6 in /usr/local/lib/python3.7/dist-packages (from raspisump) (1.6.1)
Requirement already satisfied: RPi.GPIO==0.6.5 in /usr/local/lib/python3.7/dist-packages (from hcsr04sensor>=1.6->raspisump) (0.6.5)

Am I suppose to create the folder prior in /home/pi/raspi-sump prior to executing the install?

revendoc, 2019-11-17:
Are you using sudo when running the install commands?

nickw56, 2019-11-17:
Yes. Solved that problem a while back I had missed out the step relating to versions later than Rasbian Stretch

alaudet, 2019-11-18:
Correct you need to install as follows

sudo pip install --no-binary :all: raspisump

closed 2019-11-18 by alaudet