alaudet/raspi-sump #59: Not getting email alerts or heartbeat email

nickw56, 2019-11-12:
I am using mail.smtp2go.com:2525 as my SMTP server (which I have used previously with an Arduino based monitoring project) but I am not getting any emails coming through. I have tried all combinations of  0 and 1 in the relevant bits of the conf file.  Any suggestions ?

Thanks 
Nick

alaudet, 2019-11-12:
I dont know anything about this provider.  It can be many things.  If they
use Oauth2 that can be a problem for example.  Could be any number of
things.

nickw56, 2019-11-13:
Can you tell me where the code that connects to the email server is located so I can have a look and see if I can figure out what's different ?  I'd also like to modify the code to copy the data and charts to Google Drive.  Any further information you can provide would be most helpful

alaudet, 2019-11-13:
The code for alerts is in alerts.py.

nickw56, 2019-11-16:
Hi Al,  Sorry to bother you again. Can you tell me how to unlock the alerts.py script ? I would like to try to modify it to send Alerts via IFTTT instead of email.

alaudet, 2019-11-17:
Wouldn't it just be easier to create a throw away gmail account instead?  I
don't use my everyday account to send email.

nickw56, 2019-11-17:
I've tried using my Gmail, BTInternet (yahoo) and SMTP2GO accounts so I suspect that there is a problem somewhere in my PI set up. Is there a log somewhere I can look at to see where it is failing? I would like to get the email alerts working as it really bugs me when things don't work as designed. 
However I'd also like to get the IFTTT Notification option going as this method will give me a Notification without the need to open and check the email account.  I already use this method for other types of alarm.  I have figured out how to change the permissions so I can edit the alerts.py script.

revendoc, 2019-11-17:
I can confirm that this does work as designed with gmail smtp. Recheck your configuration settings and look up using gmail over smtp as you may have to adjust some of your accounts security settings. Check for log files in /var/log/ to see if you can find any related errors.

alaudet, 2019-11-18:
If  you want to change the code i recommend forking the repository and modifying the files locally on your system.

Then you can install your changes locally from the repo directory with **sudo pip install -U .**  (don't forget the period)

If you are not used to working with Git and coding in general I recognize that there is a learning curve to doing that.  My point being that you shouldn't modify the actual files in the code folders.

However, if you insist you will need to modify files with sudo in /usr/local/lib.........

To find the location of the alerts.py you could start a uninstall which will list all the files with their complete paths.  just say N to uninstall.  I really don't recommend doing that though, or at the very least backup the file before you edit it.  any future updates would also overwrite the file.

alaudet, 2019-11-18:
If someone submits working code for an alternate way I could look at merging it in.  I just don't have the time to do all that myself.  Contact me before doing that though so you don't waste time writing stuff I would never merge for other reasons.  :-)   I prefer not to use third party services though and if at all possible sticking with actual python libraries, I have had people ask for twilio support for example, but I don't like the idea of having to keep on top of all these services changes to their TOS as they try to monetize their free services.  They don't mind breaking stuff on a whim.

nickw56, 2019-11-21:
Hi Al. It’s working fine now with GMail.   I got confused by the 2 factor verification Google Security feature which I had turned on thinking that this would be a requirement for less secure apps. I have another Pi project which uses Google Drive which has recently stopped syncing and I assumed that fixing that one would also enable Raspi-sump emails to work. 

Regarding using IFTTT for notifications I would still like to try this out and will give it a try when I have some time for experimenting.   

Thanks for your (and other contributors) help.  It’s a great solution to a very common problem.

closed 2019-11-21 by nickw56

alaudet, 2019-11-21:
Glad you got it going.  Cheers