alaudet/raspi-sump #60: Higher resolution graphic for wiring diagram?

geerlingguy, 2020-01-20:
I am currently working on getting the parts together to build this project (my old sump pump died a couple weeks ago, and I'm glad I heard it making a motor-burning-out noise or else my basement would've flooded this week since we've had a lot of precipitation!).

While looking through the documentation, I am impressed by the overall quality (thanks, and I can tell you've put a lot into this project!), but one thing that was a little difficult was reading the wiring diagram on this page: https://www.linuxnorth.org/raspi-sump/ — see below:

[image]

If possible, could you export a 2 or 4x larger version, and either embed that in the page (so those with retina displays can immediately benefit), or link the image to the full res version, so it is easier to see the exact pin designations?

It's already good enough for me to be able to build based on the pin locations, but it would be even easier if I could read the labels like Gnd, Echo, Trig, Vcc, etc.

alaudet, 2020-01-22:
I did this in Visio a long time ago.  I'll see what I can do.

alaudet, 2020-03-14:
I don't have visio anymore to modify this.  I would be happy to change the image if someone contributed a more readable version.
thanks.

alaudet, 2020-04-24:
Is this easier to read?

[image]

geerlingguy, 2020-04-24:
@alaudet - Beautiful! I don't have to squint at all now.

alaudet, 2020-04-24:
Not without making the HCSR04 picture bigger but I labelled each wire with the corresponding label on the sensor.

alaudet, 2020-04-24:
Website updated with clearer diagram

closed 2020-04-24 by alaudet

alaudet, 2020-04-24:
Thanks for the feedback...sorry it took so long.  :-)

geerlingguy, 2020-04-24:
@alaudet - Would it surprise you to know that I've had the HC-SR04 sitting on my desk about 6" from my keyboard for a couple months now? I finished putting together three of the wires but was missing a crimp-on connector for the fourth, so I ordered a bunch. They arrived, I put the box on the other side of my desk, and it's all just sitting there staring at me every day asking "when you going to monitor your sump pit!?"

It happens 😆

alaudet, 2020-04-24:
All this social distancing is giving people time to do these types of this.   :)