alaudet/raspi-sump #61: Email alerts have stopped working

nickw56, 2020-01-20:
My email alerts have stopped working. I am still getting the heartbeat notifications and the system appears to be working normally in all other respects.  Can you suggest any trouble shooting steps I can take?

alaudet, 2020-01-20:
Without more information it is hard to say.

What have you tried to generate an alert?
Is it taking readings?

nickw56, 2020-01-22:
Hi Al

My alert limit was set at 40cm but I recently had an incident when the float switch on the pump got stuck and the water level rose above 40 - no alert generated. I fixed the pump issue and then set the level to 30 cm. The pump usually runs at ~35 cm so it should be generating alerts several times a day. it is taking readings - see attached image  I do have a back-up alarm system so it's not a huge problem - just a bit puzzling 

[image]

alaudet, 2020-01-23:
I find it odd that it sends you heartbeats but not alerts.  Its the same code.  Any chance you can email me your raspisump.conf file? 

Make sure to remove your email password before sending me the file.

Also check /home/pi/logs.  Check the error_log and alert_log  Does  it think it is sending the alert?

Are you sending sms emails or just regular email?  If regular is the spam filter triggering?

I am at a loss as to why you would get a heartbeat notifcation but not an alert.

nickw56, 2020-01-29:
Here's my conf file contents as discussed
# Configuration file for Raspi-Sump
# All configurations should be done in this file.
# This file should be located in the /home/pi/raspi-sump directory

[gpio_pins]

# Raspi-Sump uses the actual GPIO numbers and not the Pi numerical pin 
# numbers.  You can use any available GPIO's for Trig and Echo.

# Trig pin sends the signal.  Value is the number of the GPIO 
# on the RaspberryPi
trig_pin = 17

# Echo pin receives the signal.  Value is the number of the GPIO 
# on the RaspberryPi
echo_pin = 27

[pit]

# Unit of measure.  You may choose to use metric or imperial measurements.  
# The unit variable must be set as metric or imperial and is case sensitive.
# Any other setting will cause an error.
unit = metric

# critical_water_level is the water level in the sump pit at which point an 
# email alert should be triggered. 
# If unit = metric then critical_water_level = centimeters
# If unit = imperial then critical_water_level = inches
critical_water_level = 30

# The pit_depth is the distance between the sensor and the bottom of the 
# sump pit.
# Like critical_water_level, enter centimeters for metric and inches for
# imperial
pit_depth = 47

# rsump.py should be run with a cron job.  However cron is limited to
# 1 minute as its minimum interval. If you need to monitor the sump level
# more frequently you can set reading_interval to the desired time in seconds.
# Default is 0 meaning only one reading will occur and the program will exit.
# If set to 0 call the script with a cron job and select your interval there.
# If setting to a value other than 0 you should run rsumpmonitor.py from a cron
# job at a regularly interval to monitor the health of the rsump.py process.
# rsumpmonitor.py is part of the raspisump software.  See documentation for
# details.
reading_interval = 0

# The temperature variable will adjust the speed of sound to reflect 
# the normal temperature of your basement.  
# If unit = metric then enter the temperature in Degrees Celcius.
# If unit = imperial then enter the temperature in Degrees Fahrenheit.
temperature = 5

# Raspi-Sump can also be used in situations where you would want to check if
# the water_level is too low. Sump pits are concerned with high water levels
# but a drinking water cistern would require an alert when it is low.  For that
# reason you can set the following value to 'low'.  Leave this value as is unless
# you specifically want alerts when the water drops below a certain level.
alert_when = high

[charts]
# This section deals with charting
# line_color takes a hex color value.
# e.g
# red(ff0000), black(000000), blue(0000ff), default rsumporange(FB921D)
line_color = FB921D

[email]

# ***Important information for Gmail users only.***
# ***Please read this thread on allowing applications to send
# ***email from Googles Gmail service which have implemented ouath2.
# ***https://github.com/alaudet/raspi-sump/issues/6

# ***Raspi-sump does not implement oauth2 for authentication so you will need 
# ***to use a workaround for gmail to send alerts.

# Set an appropriate interval for alerts.  If you are taking readings at short
# intervals it is possible to bombard yourself with SMS text messages when the
# water reaches a critical level.  For example if you are taking a reading
# every minute or less an SMS Email alert will be sent each time the reading is
# taken.  By setting the alert_interval this allows you to receive an email
# alert at a more appropriate time interval.  Readings will continue and will
# be logged but SMS text alerts will only be sent as you define it.  
# alert_interval is set in minutes. The default is 5 minutes if this value is not present.
alert_interval = 5 

# SMTP Server requires authentication (0=No, 1=Yes)
# If using localhost or possibly your ISP email this value
# will most likely be 0.  Check your ISP's configuration docs.
# For Gmail set this to 1 and fill out the corresponding 
# username and password values in this file.
smtp_authentication = 1

# SMTP Server uses TLS (0=No, 1=Yes)
# If using localhost or possibly your ISP email this value
# will most likely be 0.  Check your ISP's configuratin docs.
# For Gmail set this to 1.
smtp_tls = 1

# If server requires authentication enter username and password.
# For Gmail, activate these values with your gmail username and password.
username = 
password =

# SMTP Server and Port
# example --  smtp_server = smtp.gmail.com:587
# if using a SMTP server on the Pi use localhost:25
smtp_server = smtp.gmail.com:587
# Notifications will be sent to the following address
# example -- email_to = cellnumber@wireless_carrier (for sms email alerts)
# You can also use a regular email address
# To add multiple recipients seperate email addresses with a comma.
# e.g. Adding a single recipient
#   email_to = single_email@somewhere.com
#
# e.g. Adding multiple recipients
#   email_to = recipient1@somewhere.com, recipient2@somewhere.com
email_to =*****************@gmail.com

# Notification will be coming from the following address.
# You can enter your email address here as the sender.
# example  email_from = Raspi-Sump <youremail@yourprovider.com>
email_from = Raspi-Sump <*************@gmail.com>

# Set a heartbeat sms or email interval in order to regularly test that your
# notifications are working as intended.
# 0 = No notifications
# 1 = Send notifications
heartbeat = 1

# Set the frequency of the sms/email heartbeat notifications.
# Values can be set to any number and are in minutes.
# For reference;
# daily   = 1440 minutes
# weekly  = 10080 minutes
# Monthly = 43200 minutes
heartbeat_interval = 10080

nickw56, 2020-01-29:
Apologies for the huge font I don't know how that happened !

lecrote, 2020-02-08:
I have the same problem.
If i simulate an alert the reading on the LCD change but no email and the buzzer doesn't sound and no entry in the alert log
I see the change in the csv
In the error log I got this
2020-02-05 12:13:35,**ERROR - Signal not received. Possible cable or sensor problem.
in the process log It seem the process is stopping multiple time
2020-01-11 11:08:03,Process stopped, restarting
2020-01-11 15:54:03,Process stopped, restarting
2020-01-15 23:46:02,Process stopped, restarting
2020-01-28 10:52:03,Process stopped, restarting
2020-02-01 00:36:03,Process stopped, restarting
2020-02-01 16:12:02,Process stopped, restarting
2020-02-01 19:00:03,Process stopped, restarting
2020-02-03 08:02:02,Process stopped, restarting
2020-02-05 12:14:02,Process stopped, restarting
2020-02-08 09:40:03,Process stopped, restarting
``
Thanks
Dan

alaudet, 2020-02-08:
Are you getting heartbeat notifications?

nickw56, 2020-02-08:
Sorry but I don't think its the same problem as mine.  I am not seeing anything in the error log.  I am still getting heartbeat emails just not alerts.

lecrote, 2020-02-09:
> Are you getting heartbeat notifications?
> […](#)

In the raspisump.conf I don't see heartbeat config. I may have an old version.

alaudet, 2020-02-09:
Make sure you are using the latest version of raspisump.  Backup your
config file and replace it with the latest one manually.  Reenter all of
your values in the new file.  Test with heartbeat notifications.  If they
work simulate an actual alert.  Reassess the issue then.

lecrote, 2020-02-09:
I did find the error! 
Gmail was blocking email connection.
I unlock it and it is working again.
I will upgrade to the new version for sure.
And I would like to add on the raspberry a freezer temp sensor to.

Thanks

Dan

alaudet, 2020-02-09:
Great glad you got it working

closed 2020-03-14 by alaudet