alaudet/raspi-sump #62: create-mime.assign.pl was apparently renamed to create-mime.conf.pl

seanm, 2020-03-13:
lighttpd was not starting for me.  Manually starting revealed:

```
pi@raspberrypi:~ $ lighttpd -f /etc/lighttpd/lighttpd.conf
/bin/bash: /usr/share/lighttpd/create-mime.assign.pl: No such file or directory
```

Looking at the file system:

```
pi@raspberrypi:~ $ ls -l /usr/share/lighttpd/
total 20
-rwxr-xr-x 1 root root 6139 Jan 27  2019 create-mime.conf.pl
-rwxr-xr-x 1 root root  462 Jan 28  2019 include-conf-enabled.pl
-rw-r--r-- 1 root root 3388 Jan 28  2019 index.html
-rwxr-xr-x 1 root root  247 Jan 28  2019 use-ipv6.pl
```

Seems the file got renamed. I'll submit a patch.

alaudet, 2020-03-14:
I'll need to look into this a bit more, I have never run into that error.

What version of raspbian are you using so I can try and replicate the issue.

alaudet, 2020-03-27:
Affects version lighttpd on Raspbian Buster only.    They renamed they file create-mime.assign.pl to create-mime.conf.pl

A temporary workaround is to create a symlink as follows.  I will look at a more permanent solution that does not kill the functionality for users on Stretch and Jessie.

cd /usr/share/lighttpd
sudo ln -s create-mime.conf.pl create-mime.assign.pl
sudo /etc/init.d/lighttpd stop
sudo /etc/init.d/lighttpd start

This simply creates a soft link to the renamed file with the old name for now.

closed 2021-01-21 by alaudet

jtmoon79, 2022-10-16:
Thanks for the fix!

A little bit more info for anyone curious, this occurred to me after an upgrade from DietPi-managed Debian 9 Stretch to Debian 10 Buster, following instructions at https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

I create a PR for the dietpi docs. https://github.com/MichaIng/DietPi-Docs/pull/778