alaudet/raspi-sump #64: Allow critical_water_level to be floating point, not only integer

seanm, 2020-03-13:
I set critical_water_level to 18.5 (cm) and an exception ended up being thrown.  I believe this is because you call `getint` which probably barfs on floats.  (I'm not a python programmer, so not certain.)

I realize it's not millimetre accurate, but I think it's a nicer UI to be flexible in parsing that file.

alaudet, 2020-03-14:
I don't understand the need to have a float when there is a one cm variance in readings.  An int seems much cleaner to me in this instance.

What is the reason exactly that you need float, I don't understand exactly how **_it's a nicer UI to be flexible in parsing that file._**

seanm, 2020-03-14:
Well, the .csv file contains depth readings with one digit, ex: "18.1", so when I was trying to get email to work, I entered a value .1 less than the last reading to hopefully trigger an email. I guess I figured if it output floats, I could input floats.

>What is the reason exactly that you need float

I don't really, I just think that accepting them would do no harm, and would be one less thing for someone to debug. As a programmer myself, I figured out what the exception was telling me, but others may not have so easily.

What would be the downside in accepting floats?

PS: thanks for making and sharing this awesome tool!

alaudet, 2021-01-21:
Closing issue as I don't think changing to floats adds anything that is needed.  The readings are not mm accurate

seanm, 2021-01-21:
>The readings are not mm accurate

My motivation was not about accuracy, it was more of a "be liberal with what you accept" philosophy. A human considers `18` and `18.0` to be the same. Python doesn't, and explodes. Other programming languages do treat them the same. That config text file is basically your user interface, needing to know the subtleties of python string to number conversion is a shame.

alaudet, 2021-01-21:
I understand what you are saying.  I know the difference is between integers and floats in python.   I just didn't think you were still following the issue and since it seem minor I just closed it.    I will reopen it.  I think all it needs is to change a getint to a get which will make pit_depth a string, and then can be changed to float in water_reading().  Users could provide integers or floats in the config file that way.

I  haven't really had time to work on the program at all but if you want it now send a pull request and I am happy to add it.  Otherwise I will take a look when I start working on things again.

alaudet, 2022-02-12:
I am adding this functionality to the next major release V2.0.  No timeframe yet.

closed 2022-02-12 by alaudet