alaudet/raspi-sump #65: Spurious Peaks

nickw56, 2020-04-26:
I am getting occasional peak readings which seem to occur in the early hours of the morning. The sensor is mounted in a pipe which does not reach right to the bottom of the sump.  We have not had much rain for the last month and the the sump is empty:  I have attached some samples showing this behaviour;  

Any help you can provide to help me find the cause of this spike would be great.

[image]

[image]
 
[image]

nickw56, 2020-06-11:
Is anyone else experiencing this issue?  I have changed the sensor and it's still happening now and again.  I am certain it's not caused by the water level changing because the spike is very short (one reading) and the water level could not rise or fall that quickly.  
[image]

nickw56, 2020-06-11:
sorry closed in error

alaudet, 2020-06-15:
That is really strange, pretty consistent in the morning too.  I am not sure where to even start with that.

KevinBott, 2021-01-20:
@nickw56  Can you correlate any other activity on the Pi at this time?  Anything in Cron, etc?  I'm wondering if something else is hogging the CPU at that time.

alaudet, 2021-01-21:
This issue is now over 6 months old.  The originator can open a new request if this is still an issue.  @KevinBott  thanks for digging through old issues.  :-)

closed 2021-01-21 by alaudet