alaudet/raspi-sump #66: New install

Emork142, 2020-06-11:
My 1st install was 2 years ago.  Worked great for a year till something crashed. Couldnt figure it out so I reinstalled completely.  Was good till 5-27-2020.  .  I have something wrong but dont know where to go.  Completely stopped.  Multiple errors.  Changed sensor.  Got a new Pi3b+, reinstalled and still multiple program, errors but dont think its that.  From the sample below, can you please give me aclue where to start?

>>> %Run rsumpchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpchart.py", line 24, in <module>
    main()
  File "/usr/local/bin/rsumpchart.py", line 21, in main
    todaychart.graph(csv_file, filename, bytes2str)
  File "/usr/local/lib/python3.7/dist-packages/raspisump/todaychart.py", line 66, in graph
    csv_file, delimiter=",", unpack=True, converters={0: bytes2str}
  File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 955, in loadtxt
    fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
  File "/usr/lib/python3/dist-packages/numpy/lib/_datasource.py", line 266, in open
    return ds.open(path, mode, encoding=encoding, newline=newline)
  File "/usr/lib/python3/dist-packages/numpy/lib/_datasource.py", line 624, in open
    raise IOError("%s not found." % path)
OSError: /home/pi/raspi-sump/csv/waterlevel-20200610.csv not found.

nickw56, 2020-06-11:
My installation stopped working recently when I changed the sensor and re-set the pit depth to 34.5 cm.  I tried lots of things until I found an error indicating that the Pit Depth had to be an integer.  Changed it back to 35cm and it's working normally.  You issue is probably something different but I thought I'd let you know anyway.

alaudet, 2020-06-15:
What version of matplotlib/numpy are you running?

Try
sudo apt remove python-matplotlib python-numpy  (python version 2)

ensure python3 version is installed
sudo apt install python3-matplotlib 

that will include numpy as one of the dependencies.  

I assume you are running raspbian.

nickw56, 2020-06-15:
Hi Al 

I am on Raspian and running the newest version of matplotlib  (3.0.2-2)

But i did the re-install anyway. I noticed a bunch of errors related to openssh server but everything seems to be working ok .  

I have noticed spiders and cobwebs in the sump which is currently empty apart from an occasional pump test.  Or maybe there is something else living down there ........

Thanks again for responding I'll let you know how it goes.

Nick

[matplotlib_reinstall.txt](https://github.com/alaudet/raspi-sump/files/4782566/matplotlib_reinstall.txt)

Emork142, 2020-06-23:
Thanks. I am doing a complete rewiring. Moved the ground to a different pin
and seems to get results. Will reply when complete. I just have limited
time to work on this. Thank!

KevinBott, 2021-01-20:
@Emork142 - Did you figure out your issue?  To me it looks like the file doesn't exist yet (bottom of the stack trace)

alaudet, 2021-01-21:
As this issue has stalled I am going to close it.

closed 2021-01-21 by alaudet