alaudet/raspi-sump #68: New install on Raspbian Buster:  Webserver (lighttpd) install issues

gordee78, 2021-07-18:
Brand new Raspbian Buster install on a 3A+.

RaspiSump install itself appears to be fine (installed as 1m cron job).   But the Lighttpd installation section does not work as expected.  Webserver however, has never responded to requests.

```
sudo apt-get install lighttpd
sudo cp /home/pi/raspi-sump/sample_config/lighttpd.conf /etc/lighttpd
sudo lighttpd-enable-mod dir-listing
sudo /etc/init.d/lighttpd force-reload

```
Final command of the [install](https://github.com/alaudet/raspi-sump/blob/master/docs/install.md) is to run rsumpwebchart.py which returns:

```
pi@raspberrypi:~ $ rsumpwebchart.py
/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py:424: MatplotlibDeprecationWarning:
Passing one of 'on', 'true', 'off', 'false' as a boolean is deprecated; use an actual boolean (True/False) instead.
  warn_deprecated("2.2", "Passing one of 'on', 'true', 'off', 'false' as a "
```
This being a warning I ignored it.

Trying to investigate this a little Lighttpd says to [validate configuration file](https://redmine.lighttpd.net/projects/lighttpd/wiki/TutorialConfiguration) using the below command:

```
pi@raspberrypi:~ $ sudo cp /home/pi/raspi-sump/sample_config/lighttpd.conf /etc/lighttpd

pi@raspberrypi:~ $ lighttpd -tt -f /etc/lighttpd/lighttpd.conf                  
/bin/bash: /usr/share/lighttpd/create-mime.assign.pl: No such file or directory
2021-07-18 13:10:14: (configfile.c.1468) command "/usr/share/lighttpd/create-mime.assign.pl" exited non-zero: 127
2021-07-18 13:10:14: (configfile.c.1296) source: /etc/lighttpd/lighttpd.conf line: 29 pos: 14 parser failed somehow near here: (EOL)

```
Contents of /etc/lighttpd/lighttpd.conf

```
# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"
```
But the contents of /usr/share/lighttpd:  (Missing create-mine.assign.pl)

```
pi@raspberrypi:~ $ ls /usr/share/lighttpd/
create-mime.conf.pl  include-conf-enabled.pl  index.html  use-ipv6.pl
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $

```

alaudet, 2021-07-18:
Is it related to this issue?

https://www.linuxnorth.org/raspi-sump/faq.html#9t

gordee78, 2021-07-19:
Ack yes.   That nailed it.   Thank you.  

I actually DID do option #2 in FAQ but convoluted it with an additional change as recommended by lighttpd -tt:

```
pi@raspberrypi:~ $ lighttpd -tt -f /etc/lighttpd/lighttpd.conf

WARNING: include-conf-enabled.pl is deprecated and slated for removal.
         Replace in lighttpd.conf with:
           include "/etc/lighttpd/conf-enabled/*.conf"

2021-07-18 23:05:23: (configfile.c.1606) server.upload-dirs doesn't exist: /var/cache/lighttpd/uploads
2021-07-18 23:05:23: (mod_compress.c.275) can't stat compress.cache-dir /var/cache/lighttpd/compress/ Permission denied
2021-07-18 23:05:23: (server.c.1472) Configuration of plugins failed. Going down.
```
My apologies for this.  Thanks for helping to clear this up so quickly.

closed 2021-07-19 by gordee78