alaudet/raspi-sump #69: Raspisump reporting 0.0, but hcsr04.py reporting accurate distance

gordee78, 2021-07-28:
Utilizing brand new JSN-SR04T waterproof sensors - Revision 3.0 hardware.   Built on a new pi 3A+ on Raspian Buster.

hcsr04.py seems to be behaving very well and consistently shows good readings:
```
pi@raspberrypi:~/raspi-sump $ hcsr04.py -t 17 -e 22 -sp 0.3
trig pin = gpio 17
echo pin = gpio 22
speed = 0.3
samples = 11

The imperial distance is 22.9 inches.
The metric distance is 58.0 centimetres.

```

But rsump.py shows 0.0 over same period of time:

```
pi@raspberrypi:~/raspi-sump $ tail -f ~/raspi-sump/csv/waterlevel-20210728.csv
15:45:04,0.0
15:46:05,0.0
15:47:05,0.0
15:48:05,0.0
15:49:05,0.0
15:50:04,0.0
15:51:05,0.0
15:52:05,0.0
15:53:05,0.0
15:54:02,0.0
```

I've validated that my .conf is correctly using the proper TRIG and ECHO pins.

```
pi@raspberrypi:~/raspi-sump $ cat raspisump.conf
# Configuration file for Raspi-Sump
# All configurations should be done in this file.
# This file should be located in the /home/pi/raspi-sump directory

[gpio_pins]

# Raspi-Sump uses the actual GPIO numbers and not the Pi numerical pin
# numbers.  You can use any available GPIO's for Trig and Echo.

# Trig pin sends the signal.  Value is the number of the GPIO
# on the RaspberryPi
trig_pin = 17

# Echo pin receives the signal.  Value is the number of the GPIO
# on the RaspberryPi
echo_pin = 22
```

What else should I be checking?  Where could this be breaking between hcsr04.py and rsump.py?  Thanks

alaudet, 2021-07-29:
Depth is calculated by pit depth - distance.
Are you set to imperial or metric and what is your pit depth?

gordee78, 2021-07-29:
Pit depth is 48cm.  Metric.

I see the problem now.  All my test distances are larger than the pit depth so rsump.py is in fact operating fine.   Sorry about this - should have found this much earlier.

Happy to report that this JSN-SR04T Rev3 module seems to be very reliable and consistent.

closed 2021-07-29 by gordee78

alaudet, 2021-07-29:
All good, glad its working :-)