alaudet/raspi-sump #73: matplotlib.dates' has no attribute 'strpdate2num'

cjdock, 2021-12-28:
Graphs are not able to be created for the website.. Looks like the csv file is not able to be read even though it has the typical date and level data in it.

sudo /usr/local/bin/rsumpwebchart.py
Traceback (most recent call last):
  File "/usr/local/bin/rsumpwebchart.py", line 11, in <module>
    from raspisump import webchart
  File "/usr/local/lib/python3.9/dist-packages/raspisump/webchart.py", line 13, in <module>
    from raspisump import todaychart
  File "/usr/local/lib/python3.9/dist-packages/raspisump/todaychart.py", line 105, in <module>
    bytes2str = bytesdate2str('%H:%M:%S')
  File "/usr/local/lib/python3.9/dist-packages/raspisump/todaychart.py", line 52, in bytesdate2str
    strconverter = mdates.strpdate2num(fmt)
AttributeError: module 'matplotlib.dates' has no attribute 'strpdate2num'

alaudet, 2021-12-28:
From the python version it looks to me like you are running Raspberry Pi OS (bullseye).  I have some work to do to test on the latest version of Raspberry Pi OS.  Right now it only supports Raspbian Stretch and Buster.

Did you install matplotlib with apt or with pip?  What version are you running.

I dont have a quick solution for you as I need to install bullseye and recreate the problem.  But I guess I will start seeing this more often now.

If you absolutely need to get this going asap you may want to format another card using buster until I can test and release something that works on the latest.

cjdock, 2021-12-28:
Thanks for the fast reply!
Yes this is on bullseye. I installed matplotlib with pip.
Version matplotlib: 3.3.4
At least my alerts and notifications still work fine with your software.
Thank you for creating this software as well. I have a very high water
table and my sumps run like mad and I really like being able to know if
there is an issue before there is an issue.
Thanks!

alaudet, 2021-12-28:
Thank you very much for reporting this issue.  I am going to rename it  and leave it active until I get this fixed.  In the meantime you could always try downgrading your matplotlib version 3.0.2. which should  also install numpy 1.16.2.  

Pip can be a real hassle with matplotlib though so your mileage may vary.

cjdock, 2021-12-28:
Will do, thanks!

cjdock, 2021-12-28:
Oh, I'm incorrect on how I  installed it, I used sudo apt-get install
python3-matplotlib, not pip.

alaudet, 2021-12-29:
Issue has been recreated on a fresh install of bullseye.   An deprecated attribute in matplotlib.dates ('strpdate2num') has been removed from version 3.3.x.  This attribute handles the changing of the date format from a string so that matplotlib sees it in the proper format.

Just a matter of taking time now to fix this issue and test to make sure everything still works against buster and stretch as well.  Since Stretch is supported until the end of June 2022 the fix needs to not break anything on that version.

cjdock, 2021-12-29:
Good deal. I'll restage using Stretch for now. Its a simple process as you
know and works fine. Thanks for looking into this!

alaudet, 2021-12-29:
I recommend Buster at the very least as Stretch support ends in June of 2022.

cjdock, 2021-12-29:
Ah, ok. Will do.