alaudet/raspi-sump #74: Errors upon installing 1.3

github-t, 2022-02-16:
When I run sudo pip3 install --no-binary :all: raspisump I get the error shown below.  What am I doing wrong?

sudo pip3 install --no-binary :all: raspisump
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting raspisump
  Using cached raspisump-1.4.tar.gz (73 kB)
Collecting hcsr04sensor>=1.6
  Using cached hcsr04sensor-1.6.2.tar.gz (7.3 kB)
Collecting RPi.GPIO==0.7.0
  Using cached RPi.GPIO-0.7.0.tar.gz (30 kB)
Skipping wheel build for raspisump, due to binaries being disabled for it.
Skipping wheel build for hcsr04sensor, due to binaries being disabled for it.
Skipping wheel build for RPi.GPIO, due to binaries being disabled for it.
Installing collected packages: RPi.GPIO, hcsr04sensor, raspisump
  Attempting uninstall: RPi.GPIO
    Found existing installation: RPi.GPIO 0.7.1
    Uninstalling RPi.GPIO-0.7.1:
      Successfully uninstalled RPi.GPIO-0.7.1
    Running setup.py install for RPi.GPIO ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/setup.py'"'"'; __file__='"'"'/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_ie3tjbg/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/RPi.GPIO
         cwd: /tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/
    Complete output (87 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.9
    creating build/lib.linux-armv7l-3.9/RPi
    copying RPi/__init__.py -> build/lib.linux-armv7l-3.9/RPi
    creating build/lib.linux-armv7l-3.9/RPi/GPIO
    copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.9/RPi/GPIO
    running build_ext
    building 'RPi._GPIO' extension
    creating build/temp.linux-armv7l-3.9
    creating build/temp.linux-armv7l-3.9/source
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-armv7l-3.9/source/c_gpio.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-armv7l-3.9/source/common.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-armv7l-3.9/source/constants.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-armv7l-3.9/source/cpuinfo.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/event_gpio.c -o build/temp.linux-armv7l-3.9/source/event_gpio.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/py_gpio.c -o build/temp.linux-armv7l-3.9/source/py_gpio.o
    source/py_gpio.c: In function ‘PyInit__GPIO’:
    source/py_gpio.c:1046:4: warning: ‘PyEval_ThreadsInitialized’ is deprecated [-Wdeprecated-declarations]
     1046 |    if (!PyEval_ThreadsInitialized())
          |    ^~
    In file included from /usr/include/python3.9/Python.h:145,
                     from source/py_gpio.c:23:
    /usr/include/python3.9/ceval.h:129:36: note: declared here
      129 | Py_DEPRECATED(3.9) PyAPI_FUNC(int) PyEval_ThreadsInitialized(void);
          |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
    source/py_gpio.c:1047:7: warning: ‘PyEval_InitThreads’ is deprecated [-Wdeprecated-declarations]
     1047 |       PyEval_InitThreads();
          |       ^~~~~~~~~~~~~~~~~~
    In file included from /usr/include/python3.9/Python.h:145,
                     from source/py_gpio.c:23:
    /usr/include/python3.9/ceval.h:130:37: note: declared here
      130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
          |                                     ^~~~~~~~~~~~~~~~~~
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/py_pwm.c -o build/temp.linux-armv7l-3.9/source/py_pwm.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/soft_pwm.c -o build/temp.linux-armv7l-3.9/source/soft_pwm.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-z,relro -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.9/source/c_gpio.o build/temp.linux-armv7l-3.9/source/common.o build/temp.linux-armv7l-3.9/source/constants.o build/temp.linux-armv7l-3.9/source/cpuinfo.o build/temp.linux-armv7l-3.9/source/event_gpio.o build/temp.linux-armv7l-3.9/source/py_gpio.o build/temp.linux-armv7l-3.9/source/py_pwm.o build/temp.linux-armv7l-3.9/source/soft_pwm.o -o build/lib.linux-armv7l-3.9/RPi/_GPIO.cpython-39-arm-linux-gnueabihf.so
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/source/common.h:41: multiple definition of `module_setup'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/source/common.h:41: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/source/common.h:40: multiple definition of `setup_error'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/source/common.h:40: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/source/common.h:39: multiple definition of `rpiinfo'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/source/common.h:39:  [... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/setup.py'"'"'; __file__='"'"'/tmp/pip-install-787no5_l/rpi-gpio_c0ddf0a71e7b44a690a1a53f7bd7c433/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_ie3tjbg/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/RPi.GPIO Check the logs for full command output.

github-t, 2022-02-16:
The only way I could make this work was to do the following:
sudo pip3 install -U pip (upgrade pip to the latest)
sudo pip3 install RPi.GPIO (install current version of RPi.GPIO)
sudo pip3 install hcsr04sensor --no-deps (install current version of hcsr04sensor without dependencies because it wants an old version of RPi.GPIO)
sudo pip3 install --no-binary :all: raspisump --no-deps (install raspisump ignoring dependencies)

I'm not sure if this is the right way to do it or if there is a better way but it worked for me.  I'm interested in feedback.

alaudet, 2022-02-16:
Thank you for this information.  I think the best way forward will be to move the install of rpi-gpio to the actual apt version from the RaspberryPi.  I am going to test this.  I was pinning the old version of Rpi.GPIO to hcsr04sensor because that library has stopped supporting Raspbian Stretch which itself is still supported.

Has you installed RPi.GPIO 0.7.1 originally with pip? Or was it on the image?

Are you using Raspberry OS light version of the full desktop version?

alaudet, 2022-02-16:
Also your workaround looks sound, I really appreciate you including that.

alaudet, 2022-02-16:
ok, just tested this and Bullseye comes packaged with Rpi.GPIO 0.7.0 which is why it did not fail for me.

I uninstalled everything (raspisump, hcsr04sensor and Rpi.GPIO 0.7.0 and installed version 0.7.1 with Pip.  I then reinstalled Raspisump with my instructions and reproduced the error.

So my question is stil if you had manually installed Rpi.GPIO 0.7.1 manually with pip before hand, replacing the packaged version.  Because my install of Raspberry Pi OS (light version) comes bundled with 0.7.0 originally.

alaudet, 2022-02-16:
This is what I have reproduced as well.  I think I am going to remove the pip install requires in setup.py for Rpi.GPIO and just add install instructions for installing it with apt instead.  I already do this with matplotlib because pip was a nightmare for that.  It causes problems when new versions of Raspberry Pi OS.   On top of that non wheels installs for Rpi.GPIO just crash alltogether.

Right now my instructions work for fresh installs of Raspberry Pi OS.  But I think the following would take care of it alltogether.

First I remove the requirement for Rpi.GPIO in hcsr04sensor.  If anyone needs to use the library I can tell then to install Rpi.GPIO first

change install instructions to;

1. sudo apt install python3-rpi.gpio
2. sudo apt install --no-binary :all: raspisump

This will install raspisump and the hcsr04sensor with no problem and will address everyones use case.

I need a better way to distribute this software.

fitig, 2022-02-16:
At least in my case the current version of python3-rpi.gpio listed in the default repo is 0.7.0-0.2+b1 and I think we need 0.7.1 for this to work properly.

And yeah, I'd love to see this all in an apt package.  That seems like the cleanest mechanism given that the appropriate versions are out there.

alaudet, 2022-02-16:
works with 0.7.0-0.2+b1

I have a lot of work to do before releasing it as an apt package.  Maybe version 2.0 if ever.  I just dont have much time to devote to this project which basically does what I need it to do.

fitig, 2022-02-17:
Oh ok - mine wouldn't work unless I was 0.7.1 but maybe I had something else going on as well.

And sorry - that wasn't meant to be passive aggressive.  I'd actually like to help with getting it in an apt package.  But I suspect I have about as much time to devote to it as you. And it does what I need it to right now as well <looks outside at the flood warning with winter storm warning>.  :)

alaudet, 2022-02-17:
Oh I didn't take it that way at all.  I appreciate you having taken the time to report it.

Just to say apt was on my radar but my project structure would have to be entirely overhauled.

alaudet, 2022-02-19:
Change made

Raspisump no longer pinned to a specific version of hcsr04sensor.
In the hcsr04sensor module I removed the requirement for RPi.GPIO from setup.py.

Instructions now specifically recommend installing the apt version of RPi.GPIO 0.7.0

There are also instructions for Bullseye uses who prefer RPi.GPIO 0.7.1 to install it via Pip.  Both versions work fine on the Bullseye.

closed 2022-02-19 by alaudet