alaudet/raspi-sump #75: emailtest fails with yahoo

billotronic, 2022-09-02:
Error:

`pi@rasp-sump:~/raspi-sump $ emailtest
Traceback (most recent call last):
  File "/usr/local/bin/emailtest", line 14, in <module>
    emailtest.test_email()
  File "/usr/local/lib/python3.9/dist-packages/raspisump/emailtest.py", line 63, in test_email
    server.starttls()
  File "/usr/lib/python3.9/smtplib.py", line 774, in starttls
    self.sock = context.wrap_socket(self.sock,
  File "/usr/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "/usr/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
  File "/usr/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:1123)
`
Conf:

`# Configuration file for Raspi-Sump
# All configurations should be done in this file.
# This file should be located in the /home/pi/raspi-sump directory

[gpio_pins]

# Raspi-Sump uses the actual GPIO numbers and not the Pi numerical pin 
# numbers.  You can use any available GPIO's for Trig and Echo.

# Trig pin sends the signal.  Value is the number of the GPIO 
# on the RaspberryPi
trig_pin = 17

# Echo pin receives the signal.  Value is the number of the GPIO 
# on the RaspberryPi
echo_pin = 27

[pit]

# Unit of measure.  You may choose to use metric or imperial measurements.  
# The unit variable must be set as metric or imperial and is case sensitive.
# Any other setting will cause an error.
unit = imperial

# critical_water_level is the water level in the sump pit at which point an 
# email alert should be triggered. 
# If unit = metric then critical_water_level = centimeters
# If unit = imperial then critical_water_level = inches
critical_water_level = 18 

# The pit_depth is the distance between the sensor and the bottom of the 
# sump pit.
# Like critical_water_level, enter centimeters for metric and inches for
# imperial
pit_depth = 21

# rsump.py should be run with a cron job.  However cron is limited to
# 1 minute as its minimum interval. If you need to monitor the sump level
# more frequently you can set reading_interval to the desired time in seconds.
# Default is 0 meaning only one reading will occur and the program will exit.
# If set to 0 call the script with a cron job and select your interval there.
# If setting to a value other than 0 you should run rsumpmonitor.py from a cron
# job at a regularly interval to monitor the health of the rsump.py process.
# rsumpmonitor.py is part of the raspisump software.  See documentation for
# details.
reading_interval = 0

# The temperature variable will adjust the speed of sound to reflect 
# the normal temperature of your basement.  
# If unit = metric then enter the temperature in Degrees Celcius.
# If unit = imperial then enter the temperature in Degrees Fahrenheit.
temperature = 70

# Raspi-Sump can also be used in situations where you would want to check if
# the water_level is too low. Sump pits are concerned with high water levels
# but a drinking water cistern would require an alert when it is low.  For that
# reason you can set the following value to 'low'.  Leave this value as is unless
# you specifically want alerts when the water drops below a certain level.
alert_when = high

[charts]
# This section deals with charting
# line_color takes a hex color value.
# e.g
# red(ff0000), black(000000), blue(0000ff), default rsumporange(FB921D)
line_color = FB921D

[email]

# ***Important information for Gmail users only.***
# ***Please read this thread on allowing applications to send
# ***email from Googles Gmail service which have implemented ouath2.
# ***https://github.com/alaudet/raspi-sump/issues/6

# ***Raspi-sump does not implement oauth2 for authentication so you will need 
# ***to use a workaround for gmail to send alerts.

# Set an appropriate interval for alerts.  If you are taking readings at short
# intervals it is possible to bombard yourself with SMS text messages when the
# water reaches a critical level.  For example if you are taking a reading
# every minute or less an SMS Email alert will be sent each time the reading is
# taken.  By setting the alert_interval this allows you to receive an email
# alert at a more appropriate time interval.  Readings will continue and will
# be logged but SMS text alerts will only be sent as you define it.  
# alert_interval is set in minutes. The default is 5 minutes if this value is not present.
alert_interval = 5 

# SMTP Server requires authentication (0=No, 1=Yes)
# If using localhost or possibly your ISP email this value
# will most likely be 0.  Check your ISP's configuration docs.
# For Gmail set this to 1 and fill out the corresponding 
# username and password values in this file.
smtp_authentication = 1

# SMTP Server uses TLS (0=No, 1=Yes)
# If using localhost or possibly your ISP email this value
# will most likely be 0.  Check your ISP's configuratin docs.
# For Gmail set this to 1.
smtp_tls = 1

# If server requires authentication enter username and password.
# For Gmail, activate these values with your gmail username and password.
username = billotronic@yahoo.ie
password = xxxxxx

# SMTP Server and Port
# example --  smtp_server = smtp.gmail.com:587
# if using a SMTP server on the Pi use localhost:25
smtp_server = smtp.mail.yahoo.com:587

# Notifications will be sent to the following address
# example -- email_to = cellnumber@wireless_carrier (for sms email alerts)
# You can also use a regular email address
# To add multiple recipients seperate email addresses with a comma.
# e.g. Adding a single recipient
#   email_to = single_email@somewhere.com
#
# e.g. Adding multiple recipients
#   email_to = recipient1@somewhere.com, recipient2@somewhere.com
#   Don't forget the space after the comma in the line above.
# 
#   To test emails, run the command 'emailtest' from the pi terminal.
email_to = xxx@xxx.com

# Notification will be coming from the following address.
# You can enter your email address here as the sender.
# example  email_from = Raspi-Sump <youremail@yourprovider.com>
email_from = Raspi-Sump-Notifier <billotronic@yahoo.ie>

# Set a heartbeat sms or email interval in order to regularly test that your
# notifications are working as intended.
# 0 = No notifications
# 1 = Send notifications
heartbeat = 1

# Set the frequency of the sms/email heartbeat notifications.
# Values can be set to any number and are in minutes.
# For reference;
# daily   = 1439 minutes
# weekly  = 10079 minutes
# Monthly = 43199 minutes
heartbeat_interval = 10079`

Raspbian Bullseye on Pi Zero W, up to date. 

For the life of me I cannot seem to figure out what the problem is. Thanks for this awesome project.

alaudet, 2022-09-04:
Ok, been looking into this one and I focused on this message in the error

    ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:1123)

It appears that yahoo prefers ssl to tls for encryption.  However it should fall back to tls if ssl does not work.

Accorging to their site you should use port 465 and not 587 for your server  port.  I am not convinced it will fall back to TLS though so you will need to try it.

Please use port 465, try again and report back the error you get if it does not work.

alaudet, 2022-09-04:
Now that I read more I dont think that will work at all.  Yahoo should support port 587 tls but maybe they no longer do.  TLS still works with gmail but most likely I will need to patch my code to support ssl email connections.

alaudet, 2023-06-16:
Raspi-sump now supports ssl email as of version 1.8.1

Enable ssl in raspisump.conf by adding smtp_ssl = 1 below the smtp_tls entry and make sure to change to smtp_tls = 0

you must also use the proper port for ssl, in most cases it will be 465

e.g.

    # SMTP Server uses TLS (0=No, 1=Yes)
    smtp_tls = 0
    # SMTP Server uses SSL (0=No, 1=Yes)
    smtp_ssl = 1

    smtp_server = smtp.gmail.com:465

closed 2023-06-16 by alaudet