alaudet/raspi-sump #76: Tried running on rpi 2b- getting errors with sensor?

ormingtrude, 2023-04-25:
Having spent a few days trying to find the problems I'm having with this, I thought I'd reach out here to see if anyone has any solutions. As above I'm using an old rpi as I don't want to hook up my 4 just yet, and using the JSN sensor which I was getting distance readings using another simple Python script- GPIO 24 trig and 23 echo (FYI)

Please assume I know nothing as I'm not that far away from nothing :)

Here are the errors when running rsump.py

Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 24, in <module>
    reading.water_depth()
  File "/usr/local/lib/python3.9/dist-packages/raspisump/reading.py", line 76, in water_depth
    water_depth = water_reading()
  File "/usr/local/lib/python3.9/dist-packages/raspisump/reading.py", line 60, in water_reading
    raw_distance = value.raw_distance(sample_wait=0.3)
  File "/usr/local/lib/python3.9/dist-packages/hcsr04sensor/sensor.py", line 71, in raw_distance
    GPIO.setup(self.trig_pin, GPIO.OUT)
RuntimeError: No access to /dev/mem.  Try running as root!

alaudet, 2023-04-26:
Did you complete the full install of raspisump with instructions from the guide?

so

    sudo pip3 install --no-binary :all: raspisump

secondly, did you setup the raspisump.conf file properly in /home/pi/raspi-sump and put in all of your variables?

If it still doesnt work after that try the following two commands exactly

    sudo pip3 uninstall raspisump     (say yes to remove)

    sudo pip3 install --no-binary :all: raspisump.

You can always test the sensor manually with the following command

    hcsr04.py -t <trigpin> -e <echopin>

where trigpin and echopin are your chosen values.  I use 17 and 27 but it doenst matter.  so I would type

    hcsr04.py -t 17 -e 27

ormingtrude, 2023-04-26:
Also sorry yes I did a full install and tried on another rpi model 4 same outcome as when it’s not connected to gpio Anyways thanks for your suggestions I going to keep plugging away for nowSent from my iPhone

alaudet, 2023-05-02:
This appears to be related to the change on Raspberry Pi OS in how the default user is created.  When adding the pi user manually it does not automatically add the user to all the groups  it used to be a member of.

I have updated the documentation to add the groups (and most importantly the gpio group) when the user is created as follows

    sudo useradd -m -g users -G sudo,gpio,adm,dialout,cdrom,audio,video,plugdev,games,input,render,netdev,i2c,spi -s /bin/bash pi

If the user is already created you can use usermod to add the groups as follows;

    sudo usermod -G adm,dialout,cdrom,audio,video,plugdev,games,input,render,netdev,gpio,i2c,spi pi

alaudet, 2023-05-07:
Confirmed fix on fresh install.  Updated the install doc to add the pi user to the sudo and gpio groups if the pi user was created manually.

The first user created on the system gets gpio group added automatically and is not required if you chose pi as the first user to be created.

closed 2023-05-07 by alaudet